gin66/FastAccelStepper #210: ESP32: Why is the limit on 200kHz and not @ 500kHz?

pmneo, 2023-11-29:
I'm currently playing around with your library, but I'm asking why this is limited to 200kHz? I have adjusted StepperISR_esp32.cpp 

void StepperQueue::adjustSpeedToStepperCount(uint8_t steppers) {
  max_speed_in_ticks = 32;  // This equals 500kHz @ 16MHz
}

And this works out of the box and my oscilloscope shows a stable 500kHz signal.

So can you tell me the reason of the limit?

Thanks

gin66, 2023-11-30:
never tried and I prefer a conservative approach. besides there are some reasons (opinion):
* interrupt load gets quite high. imagine six steppers are running in parallel
* at that high speed, there will be two commands per millisecond in the queue (255 steps at 500kHz)
* every command needs minimum one interrupt with mcpwm/pcnt and approx. 16 ? with rmt. so with rmt the interrupt load is higher
* with mcpwm/pcnt, an interrupt should be ideally serviced within 2us, which is even for esp32 short. if this is not serviced, it is still not too bad. The overrun pulses will be deducted from the next command, But if the next command does not need pulses or too less (deceleration is superhigh), then the steps will be lost.
* FastAccelStepper needs to be able to work with other libs, which may have slow interrupt service routines. With high demands on the interrupt load, it may not work well with many other libs.

gin66, 2023-11-30:
Is Arduino IDE in the meantime able to define  a compile time preprocessor variable ? Last time I have checked, it was not able. Platformio has no problem with it. Besides…arduino library manager is broken for looong time now, so best to avoid Arduino IDE entirely.

An alternate solution: extend the api for esp32 only to increase the speed limit via function e.g. `stepper->setAbsoluteSpeedLimit(uint16_t max_speed_in_ticks, bool allow_unsafe_values=false)`

gin66, 2023-12-08:
included in 0.30.8

closed 2023-12-08 by gin66