gin66/FastAccelStepper #212: ESP32-S3 the first and fourth motors cannot be stopped

rqi14, 2023-12-06:
Hi. I found with an ESP32-S3 board, if there are 4 motors, the first and fourth motors cannot be stopped. It is normal for 2 and 3 motors. By "cannot be stopped", I mean by means of setting MoveTo and calling stopMove. It does not stop even after passing the target position. stopMove won't do anything to stop it. It does not change the ramp state at all.

gin66, 2023-12-06:
Thanks for testing and sorry, that you have encountered that issue. As I do not have any ESP32S3, I cannot check the functionality myself.  And looking at the code, there seems to be an issue with mcpwm/pcnt allocation and I wonder, if/how the fourth stepper actually works. I assume, later today I can push out a fix, so you can give it a try.

rqi14, 2023-12-07:
> Thanks for testing and sorry, that you have encountered that issue. As I do not have any ESP32S3, I cannot check the functionality myself. And looking at the code, there seems to be an issue with [...]

Thank you very much for the prompt fix and reply! I confirm it is working as expected now. Awesome