gin66/FastAccelStepper #220: ESP32-S3 the first and fourth motors do not move when using ESP-IDF

rqi14, 2023-12-19:
Hi. Not sure if it is related to #212, when I use ESP-IDF + arduino component, the first and fourth motors cannot be moved. The second and the third motors seem to work fine.

Also, I've identified an issue with common.h. It conflicts with common.h in WPA supplicant or something like that. It has to be renamed otherwise platformio won't compile it.

gin66, 2023-12-19:
Seems I have made a mistake and official release with the fix of #212 is not out, even I have a version tagged it with 0.30.9. Sorry for this.

Can you please share the error message ?

rqi14, 2023-12-20:
> Seems I have made a mistake and official release with the fix of #212 is not out, even I have a version tagged it with 0.30.9. Sorry for this.
> 
> Can you please share the error message ?

Hi. I feel it is like in #212. There is no error. It just doesn't move.

gin66, 2023-12-20:
In order to not chase a ghost due to version mixup, could you please clean your build directory and ensure, that you are using (and getting) 0.30.9

Perhaps I should add a readable version string to the API.

If the error persists after this fresh start, then need to continue the investigation. But still I have no esp32-s3 for test.

rqi14, 2023-12-21:
> In order to not chase a ghost due to version mixup, could you please clean your build directory and ensure, that you are using (and getting) 0.30.9
>
> Perhaps I should add a readable version string to [...]

Thanks. Cleaning up the directory works. Still, I have to rename the common.h to something else, like fas_common.h to avoid the conflict and build. Should I make this change and send a pull request later?

gin66, 2023-12-22:
you would need to change it project-wide, not only in the src/ dir.

rqi14, 2023-12-23:
> you would need to change it project-wide, not only in the src/ dir.

Sorry I don’t get your point. I want to clarify this a little. In ESP-IDF, there is a file also called common.h in the WPA supplicant module. It seems PlatformIO won’t compile because the FastAccelStepper common.h overrides it. Do you mean there is some kind of switch to change it in the project-wide level without editing the names?

closed 2024-01-04 by gin66