gin66/FastAccelStepper #223: Linear acceleration only works only on "speed 0" limits

dakcantagalli, 2024-01-06:
Hey guys.

Firstly, I would really like to thanks for the incredible work that was developed in this library!

I would like to share a strange behavior of the "setLinearAcceleration()" function.
I expected that the linear variation in acceleration would occur with every change in speed, but it only occurs at the beginning and end of the movement.
Below is a graph (speed x time) that I plotted.
[image]

In RED the behavior I expected:
[image]

If deceleration is initiated before reaching maximum speed, this is the behavior:
[image]

In RED the behavior I expected:
[image]

Am I doing something wrong or is this the expected behavior of this function?

gin66, 2024-01-06:
Good observation and the behavior like that is by design. As the doc states:

> setLinearAcceleration expects as parameter the number of steps, where the acceleration is increased linearly from **standstill** up to the configured acceleration value. If this parameter is 0, then there will be no linear acceleration phase

The background is, that this implementation is quite straightforward. The math is described [here](extras/doc/ramp_cubic_quadratic.md). Add a linear acceleration increase phase for an already running stepper requires IMHO way more sophisticated math.

gin66, 2024-03-09:
stale

closed 2024-03-09 by gin66