gin66/FastAccelStepper #226: Arduino 2.2.1, ESP32S2, example not compiling

androot2, 2024-01-12:
c:\Users\andro\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp: In member function 'void StepperQueue::startQueue_rmt()':
c:\Users\andro\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:410:26: error: no match for 'operator=' (operand types are 'volatile rmt_chndata_reg_t' and 'int')
   RMT.data_ch[channel] = 0;
                          ^
In file included from C:\Users\andro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s2/include/driver/include/driver/rmt.h:20,
                 from c:\Users\andro\Documents\Arduino\libraries\FastAccelStepper\src\common.h:134,
                 from c:\Users\andro\Documents\Arduino\libraries\FastAccelStepper\src\FastAccelStepper.h:5,
                 from c:\Users\andro\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR.h:3,
                 from c:\Users\andro\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:1:
C:\Users\andro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s2/include/soc/esp32s2/include/soc/rmt_struct.h:26:3: note: candidate: 'rmt_chndata_reg_t& rmt_chndata_reg_t::operator=(const rmt_chndata_reg_t&)'
 } rmt_chndata_reg_t;
   ^~~~~~~~~~~~~~~~~
C:\Users\andro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s2/include/soc/esp32s2/include/soc/rmt_struct.h:26:3: note:   no known conversion for argument 1 from 'int' to 'const rmt_chndata_reg_t&'
C:\Users\andro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s2/include/soc/esp32s2/include/soc/rmt_struct.h:26:3: note: candidate: 'rmt_chndata_reg_t& rmt_chndata_reg_t::operator=(rmt_chndata_reg_t&&)'
C:\Users\andro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s2/include/soc/esp32s2/include/soc/rmt_struct.h:26:3: note:   no known conversion for argument 1 from 'int' to 'rmt_chndata_reg_t&&'

exit status 1

Compilation error: exit status 1

gin66, 2024-01-12:
This happens already in an include file. Appears to me, that the arduino/espressif libs have an internal issue. Could you please start a new project and just drop these lines into it:

```
#include <driver/periph_ctrl.h>
#include <driver/pcnt.h>
#include <driver/rmt.h>
#include <soc/pcnt_reg.h>
#include <soc/pcnt_struct.h>
```

androot2, 2024-01-12:
With only included above resources it's compiling without errors.

gin66, 2024-01-12:
Thanks for the quick check. 

Apparently the issue is related to this line in FastAccelStepper: `RMT.data_ch[channel] = 0;` and that error has been already fixed [8 months ago](https://github.com/gin66/FastAccelStepper/tree/b0cef7d1037e2b88081753a5db721f10d6e29ed6). As this is already long time, I have forgotten about it and do not immediately recognize the old error.

The error was present in 0.30.0 and fixed in 0.30.1 Just the arduino library manager still delivers only 0.30.0. And apparently they are not going to fix it. Please complain [here](https://github.com/arduino/libraries-repository-engine/issues/194).

Or use something professional like [platformio](https://platformio.org/). They are [up to date](https://registry.platformio.org/libraries/gin66/FastAccelStepper)

androot2, 2024-01-12:
Thank you. I just installed 30.1 from zip file and it seems working well :)

closed 2024-01-12 by androot2

gin66, 2024-01-12:
better to use 0.30.10
Includes some more bugfixes. you can check the Changelog.md