gin66/FastAccelStepper #229: Multistage ramps

Matesxs, 2024-01-25:
Hi, thank you for pretty awesome library,
but I have one request to make.
I am working on application that will require multistage acceleration something like TMC motion controllers provide.
[image]
Obviously I could invest in this motion controller and get that functionality but that will only add cost to this device so I would like to ask if it would be possible to add similar functionality to this library.
I dont need separated accelleration and deccelleration speeds like shown on that graph I only need two ramps with different accelleration values.

gin66, 2024-01-25:
Don‘t get it. You need different values for acceleration and deceleration ? Or different acceleration values and/or deceleration values according to the ramp (speed) ? Or different acceleration for two consecutive ramps ?

In the first case, the application could just evaluate `rampState()` in the main loop. If it changes to `RAMP_STATE_ACCELERATE`, call `setAcceleration()` with the required acceleration value and followed by `applySpeedAcceleration()`. If the `rampState()` changes to `RAMP_STATE_DECELERATE`, call `setAcceleration()` with the required deceleration value and again followed by `applySpeedAcceleration()`.  Recommended is to call `setAcceleration()` once on change of ramp state and not during every main loop iteration.

For different acceleration values along the ramp, the same can be done using `getCurrentSpeedInxxx()` and the speed values, where the acceleration value shall change.

For different values of consecutive ramps, just change the acceleration value before the next ramp starts.

Matesxs, 2024-01-25:
No no, I dont need separated,only different (smaller) acceleration values at the beggining and end of the ramp.
Here is example what I would want to do:
If speed is below like 50mm/s accelleration should be 200mm/s^2 and if speed is above 50mm/s then it should be 1000mm/s^2.
Thas what I meant by multistage ramps.

gin66, 2024-01-25:
Ok. So you could use the proposed approach from the first case.

Or you look into `setLinearAcceleration()`, which allows to linearly increase the acceleration from 0 to the configured acceleration at begin and end of ramp.

Matesxs, 2024-01-29:
Thank you for all your suggestions, at the end I endup using only setLinearAcceleration, it was good enough for my application but I still think that adding support for more complex linear ramps can be good thing.
Thank you for your help :)

closed 2024-01-29 by gin66