gin66/FastAccelStepper #230: Will the library work with AC Servo Motors?

programmeddeath1, 2024-01-29:
I wanted to know if the library will work with AC Servo Motors?

I have this servo motor - https://pdf.indiamart.com/impdf/27505187955/9355230/ac-servo-motor.pdf.

It possibly runs with the same Direction/Pulse logic. Will I be able to run it with the same library?

Thanks for your help in Advance!

gin66, 2024-01-29:
FastAccelStepper does not drive any stepper directly. It still relies on a stepper driver, which translates step/dir/enable into the phased voltages to control the actual stepper motor. Glancing over the linked site, I could not find any info, that step/dir is supported by the electronics controlling that motor.

programmeddeath1, 2024-02-01:
[RS Series Servo System Start Guide.pdf](https://github.com/gin66/FastAccelStepper/files/14123851/RS.Series.Servo.System.Start.Guide.pdf)

Hi  you can check this file.

For your reference page number 5 shows the Pul/Step control terminals for the driver.

gin66, 2024-02-03:
the spec is not really clear for step interface. Is it now 24V ? must it be be a differential signal? is the step on positive or negative edge.

programmeddeath1, 2024-02-06:
[image]

It shows control through differential 5V supply. I can just confirm if its step on rising or falling edge, but the rest seems like it could work with the same setup right?

petmakris, 2024-02-06:
The Differential input 5V most probably means the inputs are opto-coupled so a floating 5V pulse between PUL+ and PUL+ will activate the input.

You can try and generate some pulses with an arduino or a square wage signal generator into them pins.
If this will work, the library will work too. 

Your question is not really relevant to this project though

gin66, 2024-02-07:
In case you can connect the PUL- to GND and PUL+ to output of the uC, then it should be ok for an avr. For an esp32, which does not output 5V, still a level shifter is needed.

programmeddeath1, 2024-02-07:
Hi yes thank you for your input guys, it should ideally work, but just wanted to confirm. I am running even the esp32 on Hybrid Drivers with 5V triggers, because usually for some 5V systems the HIGH trigger is at 2.8V and thus the esp32 with 3.3V is also able to provide the pulse triggers properly, else ofcourse I have level shifters for relay's etc which require higher min voltage more than 3.3V.

closed 2024-02-07 by programmeddeath1