gin66/FastAccelStepper #234: Pin problem with arduino mega

PregLizZz, 2024-02-23:
Hi,
I have a problem with the pins of arduino Mega and Im pretty sure I do something wrong.
This is the example code where I modified the stepPin, steps, speed, and acceleration and it works perfectly.
`#include "FastAccelStepper.h"
#include "AVRStepperPins.h" // Only required for AVR controllers

#define dirPinStepper    5
#define stepPinStepper   6

// If using an AVR device use the definitons provided in AVRStepperPins
//    stepPinStepper1A
//
// or even shorter (for 2560 the correct pin on the chosen timer is selected):
//    stepPinStepperA

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
   engine.init();
   stepper = engine.stepperConnectToPin(stepPinStepper);
   if (stepper) {
      stepper->setDirectionPin(dirPinStepper);

      stepper->setSpeedInHz(4000);       // 500 steps/s
      stepper->setAcceleration(200);    // 100 steps/s²
      stepper->move(4000);
   }
}

void loop() {
}`
When I change the pin to 7, the max speed with which it works is 1000 steps/s when I set it to 1001 or above, the motor doesn't move at all. Comparing with the docu Pin 6, 7 and 8 should work.
Can someone point me in the right direction, please?

gin66, 2024-02-23:
Known bug #221 and fixed in 0.30.10

I assume, you use ArduinoIDE with their [buggy Library manager ](https://github.com/arduino/libraries-repository-engine/issues/194), which is stuck with 0.30.0 instead of latest 0.30.11

gin66, 2024-03-09:
stale

closed 2024-03-09 by gin66