gin66/FastAccelStepper #236: Compile for Arduino UNO R4

twbrkfd1, 2024-03-03:
Compile for UNO R4 WiFi with Renesas CPU fails. The UNO R4 WiFi board has been out for quite a while now!

gin66, 2024-03-03:
[Uno R4](https://docs.arduino.cc/hardware/uno-r4-wifi/) apparently offers two uC: Renesas RA4M1 and ESP32-S3.
The  RA4M1 is not supported, but the ESP32-S3 is.
Any chance to move the stepper control to the esp32 ?

twbrkfd1, 2024-03-03:
Interesting to note is that it was Esspressif, creators of the ESP32, who ported the Arduino libraries to their CPU which created an explosion of demand! Don't see that happening at Arduino! Surely customers will want stepper motor support in the UNO R4!
Probably will abandon the UNO R4 for serious projects, and just just its 96-LED array (yes they put an array of 12x8 tiny red LEDs on it) device as warning beacon flashlight on my bicycle; Arduino used only RED LEDs!
The UNO R4, astonishingly, uses the 240MHZ ESP32 ONLY (internally clocked by a PLL from a 40Mhz XTL) as a WiFi co-processor, and only FOUR PINS are brought out as a serial port. It's one of those what-were-they-thinking designs over there in Italy (turns out Arduino has a vested interest in Renesas); like the 8086 IBM-PC  vs the Apple M68000 Mac; IBM's clout shoved the hodgepodge 8086 CPU based PC down our throats in spite of the superior fully 32 bit M68000's linear address space vs the 8086's segmented address space. I already use an ESP32-WROOM32 based UNO clone; Wemos D1 R32 with the fastAccellStepper library to control my steppers.  The WeMos is a five year old design. The WiFi is not great and has been superseded by ESP32 versions with improved WiFi radios.
The Renesas Ra4M1 (ARM Cortex-M4 core) clocks at 48MHZ.
Many libraries have STILL not been ported to the UNO R4 in spite of it being on the market for six months. There is STILL no library basic function to get the time-and-date! I just finished porting code to fetch the time-date and format it for data logging via the NTP time server. Arduino, WHY are you making us reinvent the wheel?

michel3vb, 2024-07-29:
Support for Renesas RA4M1 would be very nice! It seems that there is no other stepper library at the moment for the Renesas RA4M1

gin66, 2025-06-27:
An R4 Uno clone is in my AliExpress shopping cart waiting for sufficient content for free delivery. Even when it arrives, it will take time to start and implement. I am talking in the range of moths and not weeks. And I have still open issues with esp32p4 (#326) and raspberry pi pico (#327 - remark for setCurrentPosition).

For the expectation on performance: the RA4M1 only provides general purpose timer. Without looking into the detailed spec sheet, we are looking at one interrupt per step. Same trouble as AVR. This means for one stepper 100ksteps/s may be the limit on this device and decreasing with every further stepper. If any other interrupt - be it CAN, USB,… - takes too long, the stepper will stall. This means, even 100ksteps/s may not achievable, if paired with other peripheral drivers.