gin66/FastAccelStepper #238: ESP32 All enable pins for separate steppers on for longest stepping duration

plpers, 2024-03-04:
Per Issue #155, I am finding the same behavior as this issue, where if I have stepper1 running for 1000 steps, and stepper3 running for 5 steps, stepper3's enable pin stays on until stepper1 completes executing it's commands.

here is a code snippet below.  When i look on the oscilloscope I cant get stepper 3 enable to stop running.

#include "FastAccelStepper.h"
#include <ESP32TimerInterrupt.h>
#define dirPinStepper1 21 //24, PA2
#define enablePinStepper1 19 //22, PA0
#define stepPinStepper1 26 //26, PA4
#define dirPinStepper2 33 //24, PA2
#define enablePinStepper2 32 //22, PA0
#define stepPinStepper2 25 //26, PA4
#define stepPinStepper3 27
#define stepPinStepper4 18
#define conveyorPin 23
#define flywheelPin 22
FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper1 = NULL;
FastAccelStepper *stepper2 = NULL;
FastAccelStepper *stepper3 = NULL;
FastAccelStepper *stepper4 = NULL;

void setup() {

  engine.init();
  stepper1 = engine.stepperConnectToPin(stepPinStepper1);
  stepper2 = engine.stepperConnectToPin(stepPinStepper2);
  stepper3 = engine.stepperConnectToPin(stepPinStepper3);
  stepper4 = engine.stepperConnectToPin(stepPinStepper4);
  if (stepper1) {
    stepper1->setDirectionPin(dirPinStepper1);
    stepper1->setEnablePin(enablePinStepper1);
    stepper1->setAutoEnable(true);
    // If auto enable/disable need delays, just add (one or both):
    // stepper->setDelayToEnable(50);
    // stepper->setDelayToDisable(1000);
    stepper1->setSpeedInUs(1000);  // the parameter is us/step !!!
    stepper1->setAcceleration(1500);
    stepper1->enableOutputs();
  }
  if (stepper2) {
    stepper2->setDirectionPin(dirPinStepper2);
    stepper2->setEnablePin(enablePinStepper2);
    stepper2->setAutoEnable(true);
    // If auto enable/disable need delays, just add (one or both):
    // stepper->setDelayToEnable(50);
    // stepper->setDelayToDisable(1000);
    stepper2->setSpeedInUs(1000);  // the parameter is us/step !!!
    stepper2->setAcceleration(1500);
    stepper2->enableOutputs();
  }
  if (stepper3){
    stepper3->setEnablePin(flywheelPin);
    stepper3->setAutoEnable(true);
    stepper3->setSpeedInUs(1000);  // the parameter is us/step !!!
    stepper3->setAcceleration(1500);
    //stepper3->setDelayToDisable(1);
    //stepper3->enableOutputs();  //nuked so it doesnt boot firing and shit
  }
  if (stepper4){
    stepper4->setEnablePin(conveyorPin);
    stepper4->setAutoEnable(true);
    stepper4->setSpeedInUs(1000);  // the parameter is us/step !!!
    stepper4->setAcceleration(1500);
    //stepper4->setDelayToDisable(1);
    //stepper3->enableOutputs();  //nuked so it doesnt boot firing and shit
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  stepper3->move(20);
  stepper4->move(20);
  stepper1->move(2000);
  delay(4000);
}

plpers, 2024-03-04:
additionally, if i attempt to enable/disable outputs based on an individual stepper being stopped (e.g. stepper->isRunning) it stops all the other motors.

gin66, 2024-03-05:
Thanks for sharing the code. Before hunting any ghosts, are you by any chance using Arduino IDE with its buggy library manager ?  Arduino IDE only offers installation of 0.30.0 and there is a related bug fix implemented in 0.30..5.

plpers, 2024-03-05:
hi @gin66 - yes that works now.  Yes, I am using the arduino library manager, and it was 0.30.0.  since I'm working with kids.  I manually installed from github and it works great now!  thanks for providing a great library.

closed 2024-03-05 by plpers