gin66/FastAccelStepper #239: getCurrentPosition() returning 0

arduous09, 2024-03-05:
Hello,

I am attempting this code, but am running into issues with getCurrentPosition() returning 0 instead of the current position. Because of this, my code will not work. This is running on an ESP32s WROOM 32

`void move_motor() {
  Serial.print("Current Position: ");
  Serial.println(current_position);

  Serial.print("Moving to Position: ");
  Serial.println(move_to_step);

  stepper->setCurrentPosition(current_position);

  stalled_motor = false;

  stepper->setAcceleration(accel);
  stepper->setSpeedInHz(max_speed);
  driver.rms_current(current);
  driver.SGTHRS(stall);
  driver.TCOOLTHRS(tcools);

  if (current_position == move_to_step)
  {
    Serial.println("ALREADY THERE!");
  }
  else
  {
    Serial.println("Moving");
    stepper->moveTo(move_to_step);

    while (stepper->getCurrentPosition() != stepper->targetPos())
    {

      if (stalled_motor == true)
      {
        printf("Stalled\n");
        stepper->forceStop();
        break;
      }
      printf(".\n");
      vTaskDelay(1);
    }
  } 
  current_position = stepper->getCurrentPosition();
  printf("Motor Function Complete\n");
  Serial.print("Current Position: ");
  Serial.println(current_position);
}`

Here is an example output:
`Run Motor Function
Current Position: 0
Moving to Position: 28000
Moving
Motor Function Complete
Current Position: 0
Motor Complete`

gin66, 2024-03-05:
`moveTo()` only sets the ramp to be started, while update of the parameters take place in the cyclic stepper interrupt. This is due to the internal structure of FastAccelStepper.

Instead of:

```
while (stepper->getCurrentPosition() != stepper->targetPos())
```
use:

```
while (stepper->isRunning())
```

gin66, 2024-03-20:
stale

closed 2024-03-20 by gin66

arduous09, 2024-03-21:
This did the trick :) Thanks for your help!