gin66/FastAccelStepper #241: Execution frequency

programmeddeath1, 2024-03-21:
I am trying to run my motors for a robot through this library, I am sending the trajectory to the esp with control cycle of 0.04, which means it sends around 25 points every second. with a control cycle of 0.01, it sends 100 msgs a second or 1 traj msg every 10ms. Will the library be able to generate and execute these different velocity/acceleration/position values every 10ms and send it to the driver?

gin66, 2024-03-21:
If you send a trajectory with 10ms resolution, then the ramp generation from FastAccelStepper is not relevant. For that use case, it is better to employ the [low level interface](https://github.com/gin66/FastAccelStepper/blob/master/extras/doc/FastAccelStepper_API.md#low-level-stepper-queue-management-low-level-access).

gin66, 2024-03-21:
Answering your question: The stepper task is run every 4ms and will fill the command queue up to 20ms into the future (pre-planning). Existing commands in the queue will not be changed anymore e.g. by new values for position/speed/acceleration.

programmeddeath1, 2024-04-11:
@gin66 when you say the ramp generation from FastAccelStepper is not relevant, does it mean it will not have an effect on the 10ms executions? Is the PWM signal generation time insignificant for FastAccelStepper?

gin66, 2024-04-28:
I mean, if the trajectory is fully controlled by the application, then the ramp generator has nothing to do.

For better reaction time, #253 is added to adjust forward planning time

gin66, 2024-05-09:
Implementation released in 0.30.13

closed 2024-05-09 by gin66