gin66/FastAccelStepper #242: Rotate a fixed number of times

moebassist, 2024-03-23:
Hi,

Great library!

I’ve managed to spin my motor very quickly, in either direction. Acceleration/deceleration works great too. I can also determine the number of turns by dividing the current position by my steps/revolutions.

However, I haven’t worked out how to spin the motor a fixed number of turns….e.g.;

1. Accelerate to speed.
2. Turn xxx times
3. Decelerate when approaching xxx turns and stop upon reaching.

Any hints?

thanks!

petmakris, 2024-03-23:
You will use the methods

```
int8_t move(int32_t move, bool blocking = false);
int8_t moveTo(int32_t position, bool blocking = false);
```

the argument `move` is in number of steps and `position` is an absolute position.
To rotate the motor for say 5 rotations, you need to know you microstepping setting. Assuming 400 steps per revolution you would call `move(5 * 400, false)`

gin66, 2024-04-28:
stale

closed 2024-04-28 by gin66