gin66/FastAccelStepper #244: shift register outputs cant use certain pin numbers

plpers, 2024-03-25:
For some reason, cannot use some pin numbers.  Project still compiles but waveforms look bad on the scope, looks like pulse generator is glitching out.

Assigning EnablePin when using shift register output (I.E. PinNumber | PIN_EXTERNAL_FLAG) doesnt work for Some pins like 51, 44, 36, 67, 73, 81, 51.

Interestingly, using multiples of (8x)+1, or (8x)+2 like 41, 42, 49, 50, 65, 66 seem to work just fine.  Hypothesize something is weird with valid/invalid assignment deep inside the code.  Since I am using 12 steppers trying to come up with a list of pin numbers that dont conflict with the real output pins, have been having to do it via trial/error

plpers, 2024-03-25:
some combos work but some dont in terms of generating waveforms.  For instance:

#define dirPinStepper1 1
#define enablePinStepper1 2
#define stepPinStepper1 18  // OC1A in case of AVR

#define dirPinStepper2 3
#define enablePinStepper2 4
#define stepPinStepper2 22  // OC1B in case of AVR

works just fine.

if you add in

#define dirPinStepper3 17
#define enablePinStepper3 42
#define stepPinStepper3 19  // OC1A in case of AVR

#define dirPinStepper4 33
#define enablePinStepper4 41
#define stepPinStepper4 23  // OC1A in case of AVR

it does not work, but if you comment out 3/4 it does work.  similarly, if you swap pin numbers between 1->3 and 2->4 it stops working.

gin66, 2024-03-25:
is this a duplicate of #243 ?

plpers, 2024-03-25:
Not sure if this is a dupe of 243.  So to me there are two separate behaviors I see that are problematic - but both relate to pin assignment numbers somehow.  I went ahead and merged both sets of issues into 243 If you want to close.

closed 2024-03-27 by gin66