gin66/FastAccelStepper #245: ESP32-S3, RMT driver on 5th stepper fails

cmlpreston, 2024-03-29:
Hi,

I am finding a problem with this library when using more than 4 steppers with an ESP32-S3.

I have four steppers working (using TMC2209 driver chips and the TMC2209Stepper with UART control) correctly. 

I previously had eight working correctly with this library with an ESP32. However with the ESP32-S3 (selected for improved IO options) I find that on the 5th call to engine.stepperConnectToPin(), the following error is emitted:

(1124) rmt: rmt_isr_register(716): RMT driver installed, can not install generic ISR handler

None of steppers 5 through 8 work correctly. When movement is requested with .moveTo() the steppers just commence slow rotation and never stop.

Perhaps this is related: https://github.com/espressif/esp-idf/issues/11478 ?

I am using the Arduino IDE, and also FreeRTOS if that matters.

Thanks!

gin66, 2024-03-29:
Which other libraries/peripherals is used in the application ? Sounds more like a conflict, where another library has already installed an interrupt handler.

gin66, 2024-03-29:
and right. it is a problem, that FastAccelStepper still return a stepper instance….

cmlpreston, 2024-03-29:
It seems you were right that it was a conflict as by removing the below (and related calls) to: 

#include <WiFi.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
(Also, SPI library is loaded through the TMCStepper library).

motor #5 now works normally and no error is emitted when the stepper is configured.

Of those, I really only need the I2C Wire library and I have just re-enabled that and it seems ok, so presumable the Wifi or graphics libs were the issue. I will do some further debugging and advise if it is helpful.

gin66, 2024-04-28:
apparently no issue in FastAccelStepper besides the tracked issue 246

closed 2024-04-28 by gin66