gin66/FastAccelStepper #249: Callback when the motor starts and stop moving

ilovehotcakes, 2024-04-12:
Hi, I read through the documentation and couldn't find this information but I was wondering if it's possible to use a callback to signal to the main application when the motor starts moving and once the motor stops moving (when the PWM stops being added to the queue).

If not, what is the best way to implement it without incurring extra overhead? Thank you!

gin66, 2024-04-12:
Currently only polling can be used with `isRunning()`.

Fastest detection, if the queue is empty, is only possible in the various interrupt routines of the supported uC/derivates/implementations. This would require, that the callback is running in scope of that interrupt service. Calling a callback in this context could - especially for avr implementations - have side effects on other running steppers. Just setting a flag in the interrupt service‘s callback (in order to be fast), would again on application side require a polling mechanism. So I doubt this to be good route for implementation. 

This means, any callback could be only implemented in the cyclic 4ms task. But similar rate can be easily matched or exceeded by a polling task or main loop, or on esp32 with another cyclic task, which does the signaling or is directly initiating the expected action. So again, I do not see a good reason for that implementation.

ilovehotcakes, 2024-04-14:
Thank you for the detailed explanation.

What about when the library calls on the EN pin to turn off the standstill current from the driver to the motor? Is it possible to implement a callback so we do away with interrupts? Thank you.

gin66, 2024-04-14:
yes. There is [`setExternalCallForPin()`](https://github.com/gin66/FastAccelStepper/blob/master/extras/doc/FastAccelStepper_API.md#external-pins). Need to pay attention to short execution time of the callback.

ilovehotcakes, 2024-04-16:
Yes, I'm using that to enable/disable the current to the coils but I'm also looking to signal via semaphores to other RTOS tasks that the motor has started/stopped running. I was unsuccessful at it and I'm not sure because giving/taking the semaphores take too long. Do you have any ideas how I could make it work? Thank you.

gin66, 2024-04-18:
I am not an expert on freertos. Still I think, just setting a flag and polling in a separate task will do the trick.

ilovehotcakes, 2024-04-20:
I figured it out. I used isRunning() to poll motor's status and then signaling other tasks while it is running. Thank you for the detailed and swift replies.

closed 2024-04-20 by ilovehotcakes