gin66/FastAccelStepper #251: Build errors when using ESP core lib >=3

ChrGri, 2024-04-21:
Hi @gin66,

just a quick info, so that you are prepared. Since there is a problem with the serial output for ESP32 S3, [see](https://github.com/espressif/arduino-esp32/issues/9004), I wanted to test ESP32 core lib 3.0:
[image]

It seems to be incompaitble with this stepper lib:
[image]

BR
Chris

gin66, 2024-04-21:
Looking at [espidf](https://github.com/espressif/esp-idf/blob/636ff35b52f10e1a804a3760a5bd94e68f4b1b71/components/esp_rom/include/esp32c3/rom/gpio.h#L126), that call is present in another header. Perhaps just need another include

BTW: I have given up to understand the versions of esp32 core, idf, arduino,… Now I am only using in platformio.ini these lines:
```
[env:esp32_V6_5_0]
platform = espressif32 @ 6.5.0
```
With release versions being defined [here](https://github.com/platformio/platform-espressif32/releases)

gin66, 2024-04-28:
espressif V6.6.0 still works (and the framework even creates warnings, which is an issue for itself).

If any official espressif version breaks, then I will take this up. For now I do not plan to bugfix on 3.3.0-rc1

closed 2024-04-28 by gin66

ChrGri, 2024-04-28:
Thank you for the feedback. Following your advice, I wanted to test via platformio but I'm a bit confused since it shows me two different ESP core variants, see red lines in build log below. 

[image]

Can you confirm that indeed the 6.5 core was used and not the Arduino 2.0.14 core packake?

BR
Chris

gin66, 2024-04-29:
Apparently the 6.5.0 platform contains the framework-arduinoespressif32 v2.0.34. What in this context is “ core”, I do not know,

As I have said before, the versioning scheme is a bit confusing and I prefer to stick to the platformio-platform versioning for the build tests.