gin66/FastAccelStepper #256: [QUERY] Total step generation calculation for esp32

programmeddeath1, 2024-05-09:
Hi,

I had a few very basic queries regarding the number of steps generated per second given in the Readme.

1) The Esp32  allows up to 200000 generated steps per second. Do these steps include number of steps generated for setting position,acceleration and velocity in total at that moment? 

2) So let us say at a moment I ask the stepper to move 100mm, at 1000 mm/s and  1000 mm/s^2. It will reach in 0.447s at a max speed of 447mm/s

My steps to mm conversion gives me the following steps requirement - 

```
  stepper[num]->moveTo(2286);
  stepper[num]->setSpeedInHz(22857);
  stepper[num]->setAcceleration(22857);
```

For this motion the position and acceleration will determine the total time and velocity at any point. So does the total steps generated account for the steps needed to set position and acceleration?

Here I assume, position needs to be generated only at that instant, but the acceleration pulse generation would be constant for the entire 0.447s.

3)  In my case I am trying to run 3 stepper/servo motors using my esp32, and I wanted to know what would be the maximum Velocity and acceleration that I would be able to generate for my motions using a single esp32. My motion plan generates values as shown below in the graph. The motion is such that at specific instants it touches peak velocity of 1000mm/s to 1500mm/s. and acceleration of 30000mm/s^2. The absolute steps values for these  would be 34285 steps/s and 6857142 steps/s^2.

[image]

I am currently running a single motor with a single esp32, I want to know if a single esp32 will be able to generate these values for all three motors, for which I need to understand which parameters use up the 200000 total steps generated per second by a single esp32

gin66, 2024-05-09:
Esp32 can generate up to 200000 steps/s as a hard limit. The soft limit is the currently set speedLimit, at which the stepper will not accelerate further. These limits refers to the frequency of the generated steps or the speed of the motor. This has no relation to position and acceleration, unless the maximum speed cannot be reached. Does this clarify the first question?

For the second question: With a move of 100mm, at 1000 mm/s and 1000 mm/s^2, the max speed of 1000 mm/s will be reached in 1s. But this would need a distance larger than 100mm, so the max speed cannot be reached. This means for 100mm, the ramp generator need to concatenate acceleration and deceleration phase with same length. In other words, the motor will accelerate for 50mm and decelerate for 50mm. In order to reach 50mm with 1000mm/s², the needed time is t = sqrt(2*s/a) = sqrt(0.1s²) = 0,316s. This means, this is a different value than the 0.447s. Consequently, the motor will accelerate for 0,316, decelerate for 0,316s and the total travel time will be 0,632s. As the maximum speed is not reached, it is correct to say: _For this motion the position and acceleration will determine the total time and velocity at any point._ What is not clear the question: _So does the total steps generated account for the steps needed to set position and acceleration?_ Because the steps generated will always be in sync with the position.

For the 3rd question: The first question indicates a translation in the system of 1mm needing 22,86 steps. With 200kHz the maximum speed would be 8748 mm/s. With 1500mm/s a step speed of 34kHz is needed. If the application does not generate other heavy load on the esp32, I expect the esp32 to be able to drive all the 3 motors at 200kHz in parallel. So the 34kHz will not be a problem.

programmeddeath1, 2024-05-10:
Thank you so much for your clarification! that makes it very clear.

closed 2024-05-10 by programmeddeath1