gin66/FastAccelStepper #261: Installing from an aarch64 ubuntu system

programmeddeath1, 2024-06-01:
I am trying to install an existing fastaccelstepper code from an arduino installation on a single board chip. While compiling using the 0.30.0 version it gives the following error - 
```
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In function 'void apply_command(StepperQueue*, const queue_entry*)':
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:127:27: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'status'
  127 |   if (mcpwm->timer[timer].status.value <= 1) {  // mcpwm Timer is stopped ?
      |                           ^~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:128:25: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
  128 |     mcpwm->timer[timer].period.upmethod = 0;    // 0 = immediate update, 1 = TEZ
      |                         ^~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:138:25: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
  138 |     mcpwm->timer[timer].period.upmethod = 1;
      |                         ^~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:144:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
  144 |   mcpwm->timer[timer].period.period = ticks;
      |                       ^~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:151:12: error: 'mcpwm_dev_t' {aka 'struct mcpwm_dev_t'} has no member named 'channel'
  151 |     mcpwm->channel[timer].generator[0].utea = 1;
      |            ^~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:240:12: error: 'mcpwm_dev_t' {aka 'struct mcpwm_dev_t'} has no member named 'channel'
  240 |     mcpwm->channel[timer].generator[0].utea = 2;
      |            ^~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In function 'void init_stop(StepperQueue*)':
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:256:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'mode'
  256 |   mcpwm->timer[timer].mode.start = 0;  // 0: stop at TEZ
      |                       ^~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In function 'void mcpwm0_isr_service(void*)':
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr0_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:337:3: note: in expansion of macro 'MCPWM_SERVICE'
  337 |   MCPWM_SERVICE(MCPWM0, 0, 0);
      |   ^~~~~~~~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr1_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:338:3: note: in expansion of macro 'MCPWM_SERVICE'
  338 |   MCPWM_SERVICE(MCPWM0, 1, 1);
      |   ^~~~~~~~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr2_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:339:3: note: in expansion of macro 'MCPWM_SERVICE'
  339 |   MCPWM_SERVICE(MCPWM0, 2, 2);
      |   ^~~~~~~~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In function 'void mcpwm1_isr_service(void*)':
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr0_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:343:3: note: in expansion of macro 'MCPWM_SERVICE'
  343 |   MCPWM_SERVICE(MCPWM1, 0, 3);
      |   ^~~~~~~~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr1_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:344:3: note: in expansion of macro 'MCPWM_SERVICE'
  344 |   MCPWM_SERVICE(MCPWM1, 1, 4);
      |   ^~~~~~~~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr2_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:345:3: note: in expansion of macro 'MCPWM_SERVICE'
  345 |   MCPWM_SERVICE(MCPWM1, 2, 5);
      |   ^~~~~~~~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In member function 'void StepperQueue::init_mcpwm_pcnt(uint8_t, uint8_t)':
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:401:55: error: 'PERIPH_PWM0_MODULE' was not declared in this scope
  401 |     periph_module_enable(mcpwm_unit == MCPWM_UNIT_0 ? PERIPH_PWM0_MODULE
      |                                                       ^~~~~~~~~~~~~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:402:55: error: 'PERIPH_PWM1_MODULE' was not declared in this scope
  402 |                                                     : PERIPH_PWM1_MODULE);
      |                                                       ^~~~~~~~~~~~~~~~~~
[... 357 lines cut ...]
  369 |   rmt_set_pin(channel, RMT_MODE_TX, (gpio_num_t)_step_pin);
      |   ^~~~~~~~~~~
      |   rmt_set_gpio
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp: In member function 'void StepperQueue::disconnect_rmt()':
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:383:3: error: 'RMT' was not declared in this scope
  383 |   RMT.conf_ch[channel].conf1.idle_out_en = 0;
      |   ^~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp: In member function 'void StepperQueue::startQueue_rmt()':
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:408:3: error: 'rmt_memory_rw_rst' was not declared in this scope; did you mean 'rmt_tx_memory_reset'?
  408 |   rmt_memory_rw_rst(channel);
      |   ^~~~~~~~~~~~~~~~~
      |   rmt_tx_memory_reset
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:410:3: error: 'RMT' was not declared in this scope
  410 |   RMT.data_ch[channel] = 0;
      |   ^~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/common.h:122:43: error: 'RMT_CHANNEL_MEM' was not declared in this scope; did you mean 'RMT_CHANNEL_MAX'?
  122 | #define FAS_RMT_MEM(channel) ((uint32_t *)RMT_CHANNEL_MEM(channel))
      |                                           ^~~~~~~~~~~~~~~
/home/greenpi/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:411:19: note: in expansion of macro 'FAS_RMT_MEM'
  411 |   uint32_t *mem = FAS_RMT_MEM(channel);
      |                   ^~~~~~~~~~~
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.
```
Should installing from an aarch64 os to esp32 affect the compilation or installation of the library?

gin66, 2024-06-03:
FastAccelStepper supports up to [espressif32 @ V6.7.0](https://github.com/platformio/platform-espressif32/releases).
This includes [Arduino](https://docs.platformio.org/en/latest/frameworks/arduino.html) - v2.0.16 (based on IDF v4.4.7) and [ESP-IDF](https://docs.platformio.org/en/latest/frameworks/espidf.html) - v5.2.1

Apparently Arduino IDE has jumped to Arduino [rel 3.0.0](https://github.com/espressif/arduino-esp32/releases), which is not yet supported by platformio.

As I prefer to have an automated test environment and arduino IDE is apparently a no-go here, I will wait until platformio supports 3.0.0. From the changelog, the upgrade will be a bigger endeavour, because they have just completely reworked the RMT interface. And in the changelog, they do not mention mcpwm. But from the first message in this thread, there must be some changes in that module, too. Funny is, the driven HW is identical. So I am looking forward to just spinning the wheels...

gin66, 2024-06-04:
[https://github.com/platformio/platform-espressif32/issues/1225](issue in platformio) for v3.0.0

gin66, 2024-06-04:
Perhaps [tasmota](https://github.com/arendst/Tasmota/releases/tag/v14.1.0) could be a solution to use platformio with 3.0.0 ?

gin66, 2024-06-05:
Started branch arduino_3_3_0, which is broken

gin66, 2024-08-23:
I have now implemented support for esp-idf 5.3.0 (barely tested, but works with simple example) and only rmt with pulse counter support. no mcpwm/pcnt combo.
esp-idf up to 5.2.x will not work.
Now it is up to espressif/arduino to update to 5.3.0 as the v3.0.4 uses esp-idf 5.1
And no, I will not downgrade to support esp-idf 5.1/5.2.

This is on master

gin66, 2024-08-30:
v0.31.0 is released with support for esp-idf 5.3
and esp-idf 5.3 will be supported by arduino 3.1.0

closed 2024-08-30 by gin66