gin66/FastAccelStepper #264: Misoperation of deceleration from speeds around 2000 steps/s

sharpie7, 2024-06-16:
As mentioned in the previous issue, I am detecting a different issue with deceleration that occurs from speeds around 2000 steps/s. I used your issue 262 test program with the minSpeed and maxSpeed changed and ran it for about 8 hours on an ATMEGA168.

Output is:
```
helloNo stop iter# 21. Target: 1979.isRunning() reports true.
No stop iter# 22. Target: 1979.isRunning() reports true.
No stop iter# 23. Target: 1979.isRunning() reports true.
No stop iter# 24. Target: 1979.isRunning() reports true.
No stop iter# 25. Target: 1979.isRunning() reports true.
...
No stop iter# 201. Target: 1979.isRunning() reports true.
Speed error. Received value: 1978972. Target speed was 1979
Speed error. Received value: 1978972. Target speed was 1979
...
```
The same error sequence occured six times in the period I tested. The target values that caused the problem were either 1978 or 1979. I don't believe it occurs every time those values are used, but only sometimes.

Test program:
```
#include <FastAccelStepper.h>

#define stepPin 9
#define dirPin 10
#define USB_BAUD 250000

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

const int32_t quickAcc = 100000 * 4;
const int32_t acc = 2000;
const int32_t minSpeed = 1880;
const int32_t maxSpeed = 2000;

void setup() {
  engine.init();
  stepper = engine.stepperConnectToPin(stepPin);
  stepper->setDirectionPin(dirPin);
  Serial.begin(USB_BAUD);
  randomSeed(42);
  delay(100);
  Serial.print("hello");

}

void loop() {
  int i, count;
  long target;
 // Serial.println("Starting");
  target = random(minSpeed, maxSpeed);
  stepper->setAcceleration(acc);
  stepper->setSpeedInHz(target);
  stepper->runForward();
  while (stepper-> getCurrentSpeedInMilliHz()/1000 < target - 100)
    {};
 // Serial.println("Stopping");
  stepper ->moveByAcceleration(-quickAcc, false);
  stepper ->applySpeedAcceleration();
  count = 0;
  while (stepper -> getCurrentSpeedInMilliHz() != 0) {
    count ++;
    if (count > 20) {
      Serial.print("No stop iter# ");
      Serial.print(count);
      Serial.print(".");
      Serial.print(" Target: ");
      Serial.print(target);
      Serial.print(".");
      if (stepper -> isRunning())
        Serial.print("isRunning() reports true.");
      else
        Serial.print("isRunning() reports false.");
      Serial.println();
  }
    delay(5);
    if (count > 200)
      break;
  }
  for (i =0; i< 100; i++) {
   delay(5);
    uint32_t sp = stepper-> getCurrentSpeedInMilliHz();
    if (sp !=0) {
      Serial.print("Speed error. Received value: ");
      Serial.print(sp);
      Serial.print(". Target speed was ");
      Serial.print(target);
      Serial.println();
    }
  }
}
```

gin66, 2024-06-16:
Thanks for open a new issue. The #262 was already quite long. I assume, you are using the latest FastAccelStepper v0.30.15 with the `getCurrentSpeedInTicks()`-bugfix ? Could you please check ?

sharpie7, 2024-06-16:
Thanks. For some reason, Github is still showing me the latest version in master as 0.30.14.

gin66, 2024-06-16:
Maybe need to refresh ?
[tag](https://github.com/gin66/FastAccelStepper/tags) is up on master

sharpie7, 2024-06-17:
Refresh didn't work, but your link did.

With 0.30.15 I see no problems any more.

Thanks for quick bug-fix. I know from experience that maintenance is hard and this library does very well!

closed 2024-06-17 by sharpie7