gin66/FastAccelStepper #266: Any support for ESP32-C6 DevKits?

kyrinth021883, 2024-06-16:
Hello! New to ESP32 microcontrollers, stepper motors and drivers.  Was following some great learning examples for the TMC2209 driver by [Daniel Frenkel and his book](https://github.com/daniel-frenkel/book/tree/main).

In his TMC2209 examples he uses an ESP32 DevkitC and calls out using the FastAccelStepper library.  I am using a Sparkfun ESP32-C6 Thing Plus and am getting a "Unsupported Derivate" error when trying to compile one of the Demo sketches. 

[image]

Is the error I am getting simply due to ESP32-C6 not being supported yet by this library?  I don't see it listed on the main git page where it lists all the ESP32 platforms that have been tested/passed.

If this is the case, is there any kind of work around? Or am I just not able to use the EPS32-C6 boards I got with this library at all?  Thank you for any/all help! :)

gin66, 2024-06-16:
FastAccelStepper does not support esp32 Arduino release 3.x.x (see #261) and this will not change in a while. Good is, that esp32c6 is supported by the 2.x versions. More specifically: since platformio [6.4.0](https://github.com/platformio/platform-espressif32/releases/tag/v6.4.0). No idea how this maps to Arduino IDE versions. 

BTW: if you use Arduino IDE, you will be "blessed" with the Arduino library manager [related issue](https://github.com/arduino/library-registry/issues/2829), because Arduino IDE is stuck at FastAccelStepper 0.30.0, while lib is actually at 0.30.15.

From platform c6 support should be possible. Just until now, no one has checked, how compatible the modules are to any of the other variants.

There are now IMHO these options:
* use another stepper library
* enable C6 with either RMT or MCPWM/PCNT support and see what comes out
* wait for me to look into this and support on testing, as I have no such board/derivate

kyrinth021883, 2024-06-16:
Thank you for the fast response and info!  Initially I was looking into using platformio in Visual Studio Code, but also ran into issues where the ESP32-C6 was not supported yet, so came back to Arduino IDE.  I'll re-visit platformio also.  I am positive a lot of my issues are due to my inexperience as much of all this is new still to me and I am trying to learn as much as possible.  

I appreciate your time and efforts!

gin66, 2024-08-23:
C6 with RMT should be working with esp-idf 5.3 from theory. You can try your luck

gin66, 2024-08-30:
v0.31.0 supports esp-idf5.3.0, which will be supported by arduino core 3.1.0.
Release candidate 3.1.0-RC1 is already out.
v0.31.0 supports esp32c6

closed 2024-08-30 by gin66