gin66/FastAccelStepper #273: Library not building, examples not building : 'gpio_matrix_out' was not declared in this scope

stan69b, 2024-07-03:
Hi, i've been using your library a few month ago for a project without any issue. today i tried to get back to this project and noticed the library does not build anymore :/ I'm on a Mac M1 using arduino IDE, it used to work. now i get the following error :

`
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32.cpp: In function 'bool _esp32_attachToPulseCounter(uint8_t, FastAccelStepper*, int16_t, int16_t)':
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32.cpp:201:5: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  201 |     gpio_matrix_out(stepper->getDirectionPin(), 0x100, false, false);
      |     ^~~~~~~~~~~~~~~
      |     gpio_iomux_out
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In function 'void apply_command(StepperQueue*, const queue_entry*)':
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:127:27: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'status'
  127 |   if (mcpwm->timer[timer].status.value <= 1) {  // mcpwm Timer is stopped ?
      |                           ^~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:128:25: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
  128 |     mcpwm->timer[timer].period.upmethod = 0;    // 0 = immediate update, 1 = TEZ
      |                         ^~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:138:25: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
  138 |     mcpwm->timer[timer].period.upmethod = 1;
      |                         ^~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:144:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
  144 |   mcpwm->timer[timer].period.period = ticks;
      |                       ^~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:151:12: error: 'mcpwm_dev_t' {aka 'struct mcpwm_dev_t'} has no member named 'channel'
  151 |     mcpwm->channel[timer].generator[0].utea = 1;
      |            ^~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:240:12: error: 'mcpwm_dev_t' {aka 'struct mcpwm_dev_t'} has no member named 'channel'
  240 |     mcpwm->channel[timer].generator[0].utea = 2;
      |            ^~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In function 'void init_stop(StepperQueue*)':
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:256:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'mode'
  256 |   mcpwm->timer[timer].mode.start = 0;  // 0: stop at TEZ
      |                       ^~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In function 'void mcpwm0_isr_service(void*)':
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr0_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:337:3: note: in expansion of macro 'MCPWM_SERVICE'
  337 |   MCPWM_SERVICE(MCPWM0, 0, 0);
      |   ^~~~~~~~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr1_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:338:3: note: in expansion of macro 'MCPWM_SERVICE'
  338 |   MCPWM_SERVICE(MCPWM0, 1, 1);
      |   ^~~~~~~~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr2_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:339:3: note: in expansion of macro 'MCPWM_SERVICE'
  339 |   MCPWM_SERVICE(MCPWM0, 2, 2);
      |   ^~~~~~~~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In function 'void mcpwm1_isr_service(void*)':
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr0_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:342:3: note: in expansion of macro 'MCPWM_SERVICE'
  342 |   MCPWM_SERVICE(MCPWM1, 0, 3);
      |   ^~~~~~~~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr1_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:344:3: note: in expansion of macro 'MCPWM_SERVICE'
  344 |   MCPWM_SERVICE(MCPWM1, 1, 4);
      |   ^~~~~~~~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:315:20: error: 'volatile union mcpwm_int_st_reg_t' has no member named 'cmpr2_tea_int_st'
  315 |   if (mcpwm.int_st.cmpr##TIMER##_tea_int_st != 0) {   \
      |                    ^~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp:345:3: note: in expansion of macro 'MCPWM_SERVICE'
  345 |   MCPWM_SERVICE(MCPWM1, 2, 5);
      |   ^~~~~~~~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_mcpwm_pcnt.cpp: In member function 'void StepperQueue::init_mcpwm_pcnt(uint8_t, uint8_t)':
[... 361 lines cut ...]
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp: In member function 'void StepperQueue::disconnect_rmt()':
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:385:3: error: 'RMT' was not declared in this scope
  385 |   RMT.conf_ch[channel].conf1.idle_out_en = 0;
      |   ^~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp: In member function 'void StepperQueue::startQueue_rmt()':
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/fas_common.h:129:43: error: 'RMT_CHANNEL_MEM' was not declared in this scope; did you mean 'RMT_CHANNEL_MAX'?
  129 | #define FAS_RMT_MEM(channel) ((uint32_t *)RMT_CHANNEL_MEM(channel))
      |                                           ^~~~~~~~~~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:413:19: note: in expansion of macro 'FAS_RMT_MEM'
  413 |   uint32_t *mem = FAS_RMT_MEM(channel);
      |                   ^~~~~~~~~~~
/Users/stan/Documents/Arduino/libraries/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:495:3: error: 'RMT' was not declared in this scope
  495 |   RMT.conf_ch[channel].conf1.tx_conti_mode = 1;
      |   ^~~

exit status 1

Compilation error: exit status 1
`

I have tried reverting to an older version but the same error occurs, and i connot find any info on the internet about this.

Does anybody have an idea or a fix ? could this be due to an update from espressif somehow ? I kinda of believe it is not your lib that has an issue but have no idea where else to turn.

Thanks you for your amazing work, hope someone will have a fix :)

gin66, 2024-07-03:
same as #261 
The espressif v3.0.0 is not supported

stan69b, 2024-07-05:
Thanks for your fast answer, i wasn't sure if it was related. Should i close this issue in favor of the #261 ?

closed 2024-07-05 by gin66