gin66/FastAccelStepper #275: Incompatibility with the version 3.0.4 of the Arduino IDE ESP32 support

emc3las, 2024-08-08:
To compile a sketch using DOIT ESP32 DEVKIT V1 board and FastAccelStepper I reverted to the version 2.0.13 of the ESP32 Arduino Core.

Tobi90as, 2024-08-08:
Is this the only solution for the incompatibility issue with the 3.x Version?
Is there any solution in sight to make this library compatible with ESP Platform Version 3.x? I would really appreciate it, if I could use this library in the future as well, as it is by far the most advanced library for high speed stepper control on an ESP32.

gin66, 2024-08-08:
@emc3las That‘s great solution. Thanks for this

@Tobi90as issue #261 is dealing with the v3.x incompatibility.

Tobi90as, 2024-08-09:
Thank you very much for the quick response. I can totally understand that.
I'm afraid I can't make a real contribution, because unfortunately this is far beyond my level of understanding. I'm already happy if I can realise a few small projects with the help of great libraries like this one.
But maybe someone else will feel able to help out here.

astorun, 2024-08-22:
i really appreciate if this library is ported to v3.x in a timely manner since my project mostly depends on this library and arduino core v3!

gin66, 2024-08-23:
I have now implemented support for esp-idf 5.3.0 (barely tested, but works with simple example) and only rmt with pulse counter support. no mcpwm/pcnt combo.
esp-idf up to 5.2.x will not work.
Now it is up to espressif/arduino to update to 5.3.0 as the v3.0.4 uses esp-idf 5.1
And no, I will not downgrade to support esp-idf 5.1/5.2.

astorun, 2024-08-23:
> I have now implemented support for esp-idf 5.3.0 (barely tested, but works with simple example) and only rmt with pulse counter support. no mcpwm/pcnt combo. esp-idf up to 5.2.x will not work. Now [...]

That complicates things, at least on my case. Any specific reason why is it limited with 5.3.0? I am using esp-idf and arduino via https://github.com/pioarduino/platform-espressif32 and currently it is only supported up to 5.1.4 idf version.

Regardless, thanks for the update.

gin66, 2024-08-23:
the new rmt support prevents access to the rmt interrupts, so the whole scheme has to be adapted to the espressif opinionated way of their driver. Now an encoder is needed, which translates into the needed input for the rmt module and there are few choices, thereof only one is reasonable. That encoder is created by rmt_new_simple_encoder() and this function is not available in older idf releases.

Actually I have started to write an own encoder, because from the docs it has seemed, this is possible. Until I have found out, that a key structure is defined privately in the rmt driver….cool.

And there is another annoying espressif opinionated implementation for the pulse counter. If the pulse counter has been obtained from the driver, there is no way to find out the underlying hardware information. This underlying information is needed, to hook the pulse counter to any gpio pin or - what I need - to the rmt output pin. Solution is a hack to define just the needed espressif private struct to derive that information.

Overall the espressif opinionated 5.x esp-idf is very much limiting, the hardware possibilities of the esp32 products.

gin66, 2024-08-23:
in regard to the pulse counter: (https://github.com/espressif/esp-idf/issues/14431)

Tobi90as, 2024-08-25:
First of all, thank you for taking the time to look into the problem. Since I'm not really familiar with the technical details in the background: I can't use your new version in the current Arduino IDE with the esp32 platform version 3.0.3 at the moment, can I? Did I understand correctly that it is now up to Espressif to make your solution work at all?

Tobi90as, 2024-08-26:
This sounds promising. Thanks for the insight.

gin66, 2024-08-30:
arduino v3.1.0-RC1 is supporting esp-idf 5.3.0

closed 2024-08-30 by gin66