gin66/FastAccelStepper #276: Trouble Running Two Motors w/ a Joystick

jhneroo, 2024-08-13:
Arduino UNO
FAS version 0.30.14 from Arduino Library Manager

Hey! I'm working on a project where I'm using two stepper motors to control a large telescope (one motor at the base controlling the azimuth, and another controlling the elevation). I have this code which takes in input from an analog joystick to control the speed of both motors live with the built-in acceleration of your library.

Unfortunately, I'm running into an issue where if both stepper motors have their `.runForward()` method called at the same time, neither move. It works perfectly in all other combinations of directions (so it works when both motors are running backward, and when either motor is running forward while the other is running backward). 

I would appreciate any thoughts on why this might be happening, how to better write my code, or especially advice on how to run diagnostic testing on my system. I've tried simpler code to just test the motors and it seems like the issue is always when the `.runForward()` commands for both motors are run at once. 

```
#include "FastAccelStepper.h"

// stepperX output pins
#define xDirPinStepper 3
#define xEnablePinStepper 4
#define xStepPinStepper 9

// stepperY output pins
#define yDirPinStepper 6
#define yEnablePinStepper 7
#define yStepPinStepper 10

// joystick input pins
const int VRxPin = A0;
const int VRyPin = A1;

// deadzone radius
int dead = 5;

// joystick center values
int xCen, yCen;

// max speed and acceleration (we tested safe maximum of 30,000 steps per second, which is roughly 37.5 rps or 2250 rpm)
int maxSpeed = 300;
int maxAcceleration = 500;

FastAccelStepperEngine engine = FastAccelStepperEngine();

FastAccelStepper* stepperX;
FastAccelStepper* stepperY;

void setup() {
  Serial.begin(9600);
  // calls function to initialize the joystick center to account for noise
  initializeCenter();
  // initializes the engine
  engine.init();

  // connect the steppers to their respective pins
  stepperX = engine.stepperConnectToPin(xStepPinStepper);
    stepperX->setDirectionPin(xDirPinStepper);
    stepperX->setEnablePin(xEnablePinStepper);
    stepperX->setAutoEnable(true);
  stepperY = engine.stepperConnectToPin(yStepPinStepper);
    stepperY->setDirectionPin(yDirPinStepper);
    stepperY->setEnablePin(yEnablePinStepper);
    stepperY->setAutoEnable(true);

  stepperX->setAcceleration(maxAcceleration);
  stepperY->setAcceleration(maxAcceleration);
}

void loop() {
  // control for stepperX (elevation)
  if (analogRead(VRxPin) > xCen + dead) {
    stepperX->runForward();
    stepperX->setSpeedInHz(map(analogRead(VRxPin), xCen, 1023, 0, maxSpeed));
  } else if (analogRead(VRxPin) < xCen - dead) {
    stepperX->runBackward();
    stepperX->setSpeedInHz(map(analogRead(VRxPin), 0, xCen, maxSpeed, 0));
  } else {
    stepperX->stopMove();
  }

  // control for stepperY (azimuth)
  if (analogRead(VRyPin) > yCen + dead) {
    stepperY->runForward();
    stepperY->setSpeedInHz(map(analogRead(VRyPin), yCen, 1023, 0, maxSpeed));
  } else if (analogRead(VRyPin) < yCen - dead) {
    stepperY->runBackward();
    stepperY->setSpeedInHz(map(analogRead(VRyPin), 0, yCen, maxSpeed, 0));
  } else {
    stepperY->stopMove();
  }
  
  // "debug()" function is for all testing Serial prints. uncomment to activate
  // debug();
}

void initializeCenter() {
// set the center of the joystick to eliminate noise
  float tempX = 0;
  float tempY = 0;
  for (int i = 0; i < 50; i++) {
    tempX += analogRead(VRxPin);
    tempY += analogRead(VRyPin);
    delay(10);
  }
  xCen = tempX / 50;
  yCen = tempY / 50;
  Serial.println("All Clear.");
}

void debug() {
  Serial.print(analogRead(VRxPin));
  Serial.print(" | ");
  Serial.print(analogRead(VRyPin));
  Serial.println();
}

```

jhneroo, 2024-08-13:
Thank you so much for the reply and for optimizing my code! This makes it a bit smoother, but I'm still running into the same main issue so I'm going to assume it's a hardware issue on my end. Do you have any advice on how to test my system? I took a look at StepperDemo but don't know where to start with using it or where the documentation on that is.

gin66, 2024-08-13:
with ˋ bool test_parallel = true;ˋ the steppers do not run synchronously ?

gin66, 2024-08-13:
another test: exchange `runForward()` and `runBackward()` calls. Is the problem affecting the stepper motor other direction or does it stay the same ?

gin66, 2024-08-13:
btw: please check the ground connection between UNO and your stepper driver boards…

jhneroo, 2024-08-13:
hah. That should've been the first thing I checked — I reconnected the grounds and now it's working perfectly.

Thank you so much for your patience and helping me troubleshoot, and I'm so sorry for wasting your time! I also appreciate you and your work on the fas library since it's the only library that lets our motors go as fast as we need it to under our constraints.

closed 2024-08-14 by jhneroo