gin66/FastAccelStepper #277: Direction pin always held high even after calling runForward()/runBackward()

benpeart, 2024-08-16:
I have attached a test application using two stepper motors (left and right). When using runForward() or runBackward() the right stepper motor correctly reverses direction, but the left does not. I checked with a voltage meter and the direction pin on the left stepper is always held high. I found that if I add a call to manually pull that pin low, it will actually get pulled low and change direction correctly (caveats about only doing this while it is turning slowly as it obviously ignores any ramp down needed).

A couple of thoughts:

1. The result for the call to runForward/Backward always returns success. 
2. Since when the ramp down occurs, the pin isn't pulled high (after manually pulling it low with the call to digitalWrite) makes me think the call just isn't happening or it would pull the pin high again after I just pulled it low when the ramp down completed.
3. I have no idea why it works correctly with stepperRight but not stepperLeft

Any suggestions or ideas on how to fix or work around this anomaly would be greatly appreciated. Apologies in advance if it turns out I just have a bug in my test application.

```
        // stepper controller commands
        case 'F':
        {
            int8_t resultLeft, resultRight;
            resultLeft = stepperLeft->runForward();
            digitalWrite(stepperLeftDirPin, HIGH);
            resultRight = stepperRight->runForward();
            Serial.printf("runForward Left:%s, Right:%s\n", resultLeft ? "ERR" : "OK", resultRight ? "ERR" : "OK");
            break;
        }
            
        case 'B':
        {
            int8_t resultLeft, resultRight;
            resultLeft = stepperLeft->runBackward();
            digitalWrite(stepperLeftDirPin, LOW);
            resultRight = stepperRight->runBackward();
            Serial.printf("runBackward Left:%s, Right:%s\n", resultLeft ? "ERR" : "OK", resultRight ? "ERR" : "OK");
            break;
        }
```

[stepper-test.txt](https://github.com/user-attachments/files/16640380/stepper-test.txt)

gin66, 2024-08-16:
Apparently, the documentation is not clear enough. There must be only ONE FastAccelStepperEngine. So instead of:

```
FastAccelStepperEngine stepperEngineLeft = FastAccelStepperEngine();
FastAccelStepperEngine stepperEngineRight = FastAccelStepperEngine();

stepperLeft = stepperEngineLeft.stepperConnectToPin(stepperLeftStepPin);
stepperRight = stepperEngineRight.stepperConnectToPin(stepperRightStepPin);
```
use only:
```
FastAccelStepperEngine stepperEngine = FastAccelStepperEngine();

stepperLeft = stepperEngine.stepperConnectToPin(stepperLeftStepPin);
stepperRight = stepperEngine.stepperConnectToPin(stepperRightStepPin);
```
No idea, what side effects this has or if it works at all.

Please try this and check, if it then works as expected.
If not, please advise, which esp32 derivate/board is used.

benpeart, 2024-08-16:
Awesome! That is an easy answer. Yes, just having a single stepperEngine works correctly. Thank you.

A quick suggestion: since there should only ever be a single instance of the stepperEngine, how about making it a singleton? That could completely hide the stepperEngine and prevent people (like me) from getting it wrong.  If you aren't sure how to do that, Copilot can help:

1. Make the Constructor Private: This prevents the creation of instances from outside the class.
2. Delete the Copy Constructor and Assignment Operator: This ensures that the singleton instance cannot be copied.
3. Create a Static Method to Get the Instance: This method will return the single instance of the class.
4. Use a Static Pointer to Hold the Instance: This pointer will be used to store the single instance of the class.

closed 2024-08-16 by benpeart