gin66/FastAccelStepper #279: Compilation issue with PlatformIO and gnuC++17

RemiDeparis59, 2024-09-05:
hello Thanks a lot for this libs which is very use full, I discover an issue when I added thos 2 lines in my platform.ini 
build_unflags = -std=gnu++11
build_flags = -std=c++17

I get such compilation error, hope it will help you to perform your libs :
BR .

`Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\FastAccelStepper.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\FastAccelStepper_idf4_esp32_pcnt.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\FastAccelStepper_idf5_esp32_pcnt.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\PoorManFloat.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\RampCalculator.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\RampConstAcceleration.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\RampGenerator.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR_avr.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR_due.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR_esp32.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR_idf4_esp32_mcpwm_pcnt.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR_idf4_esp32_rmt.cpp.o
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR_idf4_esp32c3_rmt.cpp.o
In file included from D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:26,
                 from D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:26,
                 from D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/Stream.h:26,
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR_idf4_esp32s3_rmt.cpp.o
                 from D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.h:50,
                 from include/WheelCom.h:4,
Compiling .pio\build\esp32-c3-devkitc-02\libd16\FastAccelStepper\StepperISR_idf5_esp32_rmt.cpp.o
                 from src/WheelCom.cpp:1:
.pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In function 'const T* ArduinoJson::V710PB22::detail::pgm_read(const T* const*)':
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:52:16: error: expected ';' before '_addr'
   typeof(addr) _addr = (addr); \
                ^~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:53:21: error: '_addr' was not declared in this scope
   *(void * const *)(_addr); \
                     ^~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:53:21: note: suggested alternative: 'ip_addr'
.pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In function 'uint32_t ArduinoJson::V710PB22::detail::pgm_read(const uint32_t*)':
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:44:3: error: 'typeof' was not declared in this scope
   typeof(addr) _addr = (addr); \
   ^~~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:44:3: note: suggested alternative: 'feof'
Compiling .pio\build\esp32-c3-devkitc-02\lib50f\fifoRotation\FifoRotation.cpp.o
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:45:28: error: '_addr' was not declared in this scope
   *(const unsigned long *)(_addr); \
                            ^~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:45:28: note: suggested alternative: 'ip_addr'
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:46:2: error: void value not ignored as it ought to be
 })
  ^
.pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In function 'float ArduinoJson::V710PB22::detail::pgm_read(const float*)':
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:48:3: error: 'typeof' was not declared in this scope
   typeof(addr) _addr = (addr); \
   ^~~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:48:3: note: suggested alternative: 'feof'
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:49:20: error: '_addr' was not declared in this scope
   *(const float *)(_addr); \
                    ^~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:49:20: note: suggested alternative: 'ip_addr'
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:50:2: error: void value not ignored as it ought to be
 })
  ^
In file included from D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:26,
                 from D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:188,
                 from .pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson.hpp:18,
                 from .pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson.h:9,
                 from include/FBJsonDocument.h:4,
                 from src/FBJsonDocument.cpp:2:
.pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In function 'const T* ArduinoJson::V710PB22::detail::pgm_read(const T* const*)':
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:52:16: error: expected ';' before '_addr'
[... 23 lines cut ...]
                           ^
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:45:28: note: suggested alternative: 'ip_addr'
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:46:2: error: void value not ignored as it ought to be
 })
  ^
.pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In function 'float ArduinoJson::V710PB22::detail::pgm_read(const float*)':
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:48:3: error: 'typeof' was not declared in this scope
   typeof(addr) _addr = (addr); \
   ^~~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:48:3: note: suggested alternative: 'feof'
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:49:20: error: '_addr' was not declared in this scope
   *(const float *)(_addr); \
                    ^~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:49:20: note: suggested alternative: 'ip_addr'
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:50:2: error: void value not ignored as it ought to be
 })
  ^
.pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'const T* ArduinoJson::V710PB22::detail::pgm_read(const T* const*) [with T = char]':
.pio/libdeps/esp32-c3-devkitc-02/ArduinoJson/src/ArduinoJson/Deserialization/DeserializationError.hpp:85:42:   required from here
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:52:9: error: 'typeof' was not declared in this scope
   typeof(addr) _addr = (addr); \
   ~~~~~~^~~~~~
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:52:9: note: suggested alternative: 'feof'
D:/Users/xxxxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/pgmspace.h:53:27: error: void value not ignored as it ought to be
   *(void * const *)(_addr); \`

gin66, 2024-09-05:
Thanks for the info. The error is not related to FastAccelStepper. This is a problem originating in arduino-espressif framework. You could raise an issue in their github. 
Perhaps it works, if you remove ArduinoJson.

gin66, 2024-09-15:
stale

closed 2024-09-15 by gin66