gin66/FastAccelStepper #280: Motor does not stop sometimes when changing speed

githubusertoday, 2024-09-20:
I'm using a Atmega32u4@16MHz and FastAccelStepper 0.31.0 with a single stepper motor and it just works fine.

But there is one little problem: when updating the motor speed with

`stepper->setSpeedInHz()`
`stepper->applySpeedAcceleration()`

the `stopMove()` command is ignored sometimes and the motor does not stop.

I'm reading a poti ten times a second for realtime speed adjustment.
The motor is started with `stepper->runForward()`

As workaround i need to frequently check `stepper->isRunning()` and make the motor stopping.

gin66, 2024-09-20:
Thanks for reporting. I have confirmed the error by adding a failing test case. The bug is already fixed and the release for the version 0.31.1 is ongoing.

In course of this bug, I have noticed, that `stepper->isStopping()` hasn‘t worked as expected and this is fixed at the same time.

githubusertoday, 2024-09-20:
Thank you for the super-fast solution.

closed 2024-09-20 by githubusertoday