gin66/FastAccelStepper #281: Reverse moves cause drift

AdamDrefs, 2024-10-05:
I am using an Arduino Mega 2560 to control some steppers. I need it to control 2 motors on a robot with high precision and decently high speeds which regular AccelStepper couldn't handle. I am only using one stepper at the moment for testing purposes. Forward moves do fine, acting completely normal, but I noticed that when I tell the motor to move backwards, it does the move as intended but then continues rotating very slowly even after it has supposedly stopped. The code thinks it has stopped, as forceStop() and stopMove() do nothing and isRunning() returns false, but the motor continues to drift. I have tried writing HIGH to the step pin as well, but nothing has worked so far.  

Here's my code:

#include "FastAccelStepper.h"

#define microsteps 16
#define maxSpeed 4
#define maxAccel 12

// StepperL
#define dirPinStepperL 22
#define stepPinStepperL 8

//StepperR
#define dirPinStepperR 23
#define stepPinStepperR 9

//Robot Details
#define dowelOffset 20          //mm
#define wheelSeparation 208     //mm
#define wheelCircumference 250  //mm

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepperL = NULL;
FastAccelStepper *stepperR = NULL;

void setup() {
  Serial.begin(9600);
  //initialize engine
  engine.init();

  //define motor pins
  stepperL = engine.stepperConnectToPin(stepPinStepperL);
  stepperL-> setDirectionPin(dirPinStepperL);

  stepperR = engine.stepperConnectToPin(stepPinStepperR);
  stepperR-> setDirectionPin(dirPinStepperR);

  //set max speeds and accels
  stepperL->setSpeedInHz(maxSpeed * 200 * microsteps);
  stepperL->setAcceleration((int32_t)maxAccel * 200 * microsteps);
  
  stepperR->setSpeedInHz(maxSpeed * 200 * microsteps / 2);
  stepperR->setAcceleration((int32_t)maxAccel * 200 * microsteps);

  move(5000);
  //avoid negative moves, they cause drift
}
int32_t distToSteps(int32_t dist){
  //converts distance to steps
  return dist / wheelCircumference * 200 * microsteps;
}

void run(int32_t stepsL, int32_t stepsR){
  //wait for steppers to stop before executing next move
  //this allows for precalculating moves
  while(stepperL->isRunning() || stepperR->isRunning()){}

  //move the motors
  stepperL->move(stepsL);
  stepperR->move(stepsR);
}

void move(int32_t dist){
  Serial.print((int32_t)3200 * (int32_t)20);
  int32_t steps = distToSteps(dist);
  run(steps, steps);
}

void rotate(int32_t theta){
  //2πr * Θ/360
  int32_t dist = (PI * 2 * wheelSeparation/2) * (theta / 360);
  int32_t steps = distToSteps(dist);
  run(steps, -steps);
}

void loop() {}

gin66, 2024-10-05:
Hi,

should not be like this. Could you please check your ground connection from uC to the stepper controller ? And that the enable pin of the stepper controller is properly set.

Slight remark: This construct may not be calculated as expected: `return dist / wheelCircumference * 200 * microsteps;`. Any value of dist between 0 and  (wheelCircumference-1) will yield 0. If this is intended, then great.

AdamDrefs, 2024-10-06:
> **[gin66](/gin66) ** co

Grounding the driver to the uC fixed it! And yes, I saw that issue and fixed it right after this post.

closed 2024-10-06 by AdamDrefs