gin66/FastAccelStepper #283: Task Not Found ESP32

Fryedrocket, 2024-10-18:
When I upload the UsageExample demo from the library for esp32, the ESP32 does not generate any pulses and over the serial monitor I get flooded with this error:

`E (11043451) task_wdt: esp_task_wdt_reset(763): task not found`
 
I am using ESP32 and A4899 with nothing else connected. The example code from the library is also unmodified when I uploaded it. What is causing this error? Thank you

gin66, 2024-10-18:
Sorry for hearing this. Let’s if we can isolate the root cause. Could you please help and provide more details ? Do you use platformio or arduino ? Which version of espidf or espressif-arduino ? Especially if using arduino, which version of FastAccelStepper is in use ?

Fryedrocket, 2024-10-18:
Thank you for your swift reply, sorry I did not provide the details I the original post. I was using arduino ide 2.3.3 and FastAccelStapper 0.30.15. However I was only using arduino IDE because I wanted to test the functionality, if it works better on platformio, I can easily make the switch to platformio.

gin66, 2024-10-18:
Thanks for the info. The arduino IDE version does not tell me, which espressif arduino version is in use.
Could you please try FastAccelStepper v0.31.2 ? This version should be compatible with more recent espressif-arduino versions. Specifically espidf v4.x und v5.3.1. The v5.0-2 will not work. The espressif-arduino core version maps somehow to the espidf version.

If the arduino IDE does not offer you that recent version, then this is no surprise. The arduino library manager is broken and the [issue](https://github.com/arduino/libraries-repository-engine/issues/194) from May 2023 is still open. platformio allows me to run automatic builds for all espidf-versions, which is AFAIK not possible with arduino ide. Still I miss an esp32 simulator, which is available for avr.

gin66, 2024-10-18:
Good question, how to check the version. On my main computer have not bothered to even install arduino IDE anymore. So I cannot guide you here.

Fryedrocket, 2024-10-18:
I will check if I can update the library, though I think that it is the newest version available in arduino IDE. Sounds like platformio would be better anyways?

Fryedrocket, 2024-10-19:
PlatformIO builds and works as expected, thank you for you help.

closed 2024-10-19 by gin66