gin66/FastAccelStepper #285: Enabling PSRAM on esp32-s3 triggers crash when stepper starts running

leokeba, 2024-10-24:
Hi, I'm having issues using Platformio with arduino espressif32 @6.9.0 on an esp32-s3-super-mini board.

If I use this setup without the `BOARD_HAS_PSRAM` build flag, everything works. When I enable it, I get a crash as soon as I start moving the stepper. Here's the serial log ouptut :
```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x40374340  PS      : 0x00060034  A0      : 0x803793f4  A1      : 0x3fc974d0  
A2      : 0x3de00800  A3      : 0x00000001  A4      : 0x3fc9e2f0  A5      : 0x00060023  
A6      : 0x00060021  A7      : 0x00000001  A8      : 0xbad00bad  A9      : 0x00000001  
A10     : 0x00060320  A11     : 0x00000001  A12     : 0x803810f9  A13     : 0x3fcf1630  
A14     : 0x02c96058  A15     : 0x00ffffff  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0xbad00bf1  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4037433d:0x3fc974d0 0x403793f1:0x3fc974f0 0x40379f79:0x3fc97510 0x40377c9a:0x3fcf1680 0x4037a5a9:0x3fcf16a0
  #0  0x4037433d in _KernelExceptionVector at ??:?
  #1  0x3fc974d0 in port_IntStack at ??:?
  #2  0x403793f1 in shared_intr_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/intr_alloc.c:407
  #3  0x3fc974f0 in port_IntStack at ??:?
  #4  0x40379f79 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1118
  #5  0x3fc97510 in port_IntStackTop at ??:?
  #6  0x40377c9a in spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:120
  #7  0x4037a5a9 in ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_ipc/src/esp_ipc.c:77

ELF file SHA256: 711317310f56937a
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380f67
  #0  0x40380f67 in compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:301
```

After a bit of research, the only thing I found was the following build flag that seemed interesting : `CONFIG_RMT_ISR_IRAM_SAFE`

However, that didn't fix the issue and I am a bit lost as to where to look, it seems to me that all critical ISR functions are correctly prefixed with `IRAM_ATTR` but this not my best area of expertise. Does anyone have a suggestion on where to look next ?

Thanks

gin66, 2024-10-24:
Could it be, that the PSRAM interface is conflicting with Step/Dir/Emable-Pins ?

gin66, 2024-10-25:
after some study of the available info in the datasets, pin conflict may not happening with the available pins of the module.

leokeba, 2024-10-25:
PSRAM is supposed to be hooked up to pins 33-37, I am not using them for the stepper driver. Also, I tried with two different boards based on esp32-s3 with PSRAM, devkit-c and super-mini, they both show the same issue.

leokeba, 2024-10-25:
> What if you select the rmt driver on stepperConnectToPin ?
> 
> ```
> #define DRIVER_MCPWM_PCNT 0
> #define DRIVER_RMT 1
> #define DRIVER_DONT_CARE 2
>   FastAccelStepper* stepperConnectToPin(uint8_t step_pin,
>                                         uint8_t driver_type = DRIVER_DONT_CARE);
> ```

I get no crash using DRIVER_RMT, but the steppers don't run either, wether the PSRAM flag is enabled or not. It looks like they receive one step (I hear a click) when I start a move, but nothing more.

Is the RMT driver confirmed to work on esp32-s3 chips ?

leokeba, 2024-10-25:
Update : I noticed the RGB LED stopped working when trying to use DRIVER_RMT. I was using the neopixelWrite function from arduinoespressif, turns out it also uses rmt to write data to the LEDs.

Good news is that this solves the initial issue, I can build with PSRAM enabled and run the steppers. But I can't use neopixelWrite at the same time. I can live with that, there's surely a way to control the LED without rmt, but I am also interested in helping resolve both issues if you consider them bugs.

In any case, thanks for the help.

gin66, 2024-10-25:
Happy to hear, that it works with the rmt driver.

Here is the funny thing. The mcpwm/pcnt driver is a legacy driver, because I do not intend to adapt it to espidf 5.x version. Besides those guru meditations are very hard to debug, give a lot of rather useless information. And if the hardware is not with me, this would take a lot of patience on both sides for trial and error.

The espidf 5.x drops the notion, that you can directly access the hardware. That‘s the main reason, why I do not plan to port the mcpwm/pcnt driver to that new release. Due to this separation from the hardware, two software libs can access the rmt hardware without conflicts. So the neopixelWrite and FastAccelStepper will be able to use it simultaneously.

Only problem is, that my port to espidf 5.x starts working with espidf5.3 and that version is until now only in the espressif-arduino beta. So it is a matter of time until both issues gets fixed by updates. Or you avoid arduino and use espidf directly. I have a bunch of build scripts to test that it compiles and the one or other version is tested on  esp32 hw. Compatibility for all other derivatives is then up to espidf.

closed 2024-10-25 by gin66