gin66/FastAccelStepper #286: Ability to override queue for urgent actions

sharpie7, 2024-10-25:
Thanks as always for this well maintained library.

I have an application where there is a need to respond urgently to detection of end-stops so that items in rapid motion stop in the space between the end-stop detection and the limits of travel. 

The default 20ms queue buffer is too long for this application. I can use the API to reduce the length of the buffer, but even at 8ms (ie two interruputs) the delay is longer I would like.

But, it makes me think more generally about the queue management. For urgent tasks that override the pre-defined motion, would it be possible to erase the existing queue and instead load the queue with new tasks to be handled on the next interrupt? I can see this might be useful in a number of scenarios (including mine). The API doesn't currently support it, but it seems like something that could be added.

gin66, 2024-10-25:
That feature exists: `forceStopAndNewPosition(int32_t new_pos);` It stops the queue very hard and by this, the lib will loose the correct position. That”s the reason for the position parameter. If the motor has already run in the end stop, most likely it is at a known position anyway and calibration/homing will be needed.

For immediately start processing, I doubt this will be implemented. It sounds easy, but is rather complex with the current structure of the lib.

gin66, 2024-10-25:
just read again. You do not want to stop the motor, but rewrite the queue. So more or less pop the existing commands and append new ones. Difficulty is, that filling the queue and process its entries are asynchronous, and the queue fill task may be delayed in processing…

sharpie7, 2024-10-25:
> You do not want to stop the motor, but rewrite the queue. So more or less pop the existing commands and append new ones.

Yes, exactly. Specifically, I want to force the motor to immediately start decelerating from its current speed using a deceleration rate I give it.

> Difficulty is, that filling the queue and process its entries are asynchronous, and the queue fill task may be delayed in processing. FastAccelStepper would need to guess, how many commands are allowed to be popped without running in the problem, that the queue is out of commands…

Yes, I forsee this problem. My idea is something like this:
1) The application writes a new queue in a separate buffer. This could be done using the existing API, but with a qualifier to indicate the events are not going on the main queue.
2) Once the new queue is created the application calls a function that atomically (e.g. with interrupts off) removes any umprocessed events from the main queue and replaces it with the new queue.

That way the process can continue to operate normally until there is enough new buffered data to allow operation without exhausing the queue.

There are other ways you might do it, but that was the first approch I thought of. I suspect the common use of this requirement is to stop or change direction using a specified deceleration. It might be easier to just create a solution for that case.

gin66, 2024-10-26:
Thanks for the proposal. Popping/Removing commands from the queue end or even switching to another queue is in regard to the ramp generator still a major endeavor. The internal state of the ramp generator is linked to the content of the queue. This means, modification of the queue would mean to update the ramp generator internal state. For slow moves, the last commands may be only pauses and then it gets even more tricky.

BTW: Have you tried to use `setForwardPlanningTimeInMs(uint8_t ms)` ? Perhaps in your setup (low uC load) a parameter of 5 may work.

sharpie7, 2024-10-26:
Thanks for the consideration and feedback. 

I think that for this application I need to optimize the treatment of this scenario as much as possible if it is going to perform how I want. I don't see any existing libs that cater well for all the requirements. Realistically it looks like I am going to have to write my own stepper pulse generator and ramps. That's the way it goes sometimes!

closed 2024-10-26 by sharpie7