gin66/FastAccelStepper #288: How high can the control frequency be on the esp32 using this library?

programmeddeath1, 2024-10-27:
I have a esp32 controlling 3 AC Servo motors of a delta robot with a maximum RPM of 3000 and an entire motion plan sent to the esp32 over UDP at a rate 1.12ms per message.

I am trying to execute motions on the esp32 using FastAccelStepper at two different control cycles - 0.02 (ie a point every 20ms) and 0.015( a point every 15ms) run at a max velocity of 1500 RPM. The motion plan for the two control cycles for the same positions is as follows - 

```
CC = 0.02
total points :  47
POINT: 
 positions: [0.26831230272552986, 0.26831230272552986, 0.0029724586491534327]
velocities: [-0.5669126948666714, -0.5669126948666714, 0.07275552637463709]
accelerations: [-21.37, -21.37, 0.0]
effort: [0.0]
time_from_start: 
  secs: 0
  nsecs:  26528437
Count :  0
POINT: 
 positions: [0.24575334131226417, 0.24575334131226417, 0.004902549033968065]
velocities: [-1.1338253897333428, -1.1338253897333428, 0.07275552637463709]
accelerations: [-21.37, -21.37, 0.0]
effort: [0.0]
time_from_start: 
  secs: 0
  nsecs:  53056874
Count :  1
POINT: 
 positions: [0.21191489919236567, 0.21191489919236567, 0.0068326394187826975]
velocities: [-1.133825389733343, -1.133825389733343, 0.07275552637463709]
accelerations: [21.37, 21.37, 0.0]
effort: [0.0]
time_from_start: 
  secs: 0
  nsecs:  79585310
Count :  2
POINT: 
 positions: [0.18935593777909998, 0.18935593777909998, 0.00876272980359733]
velocities: [-0.5669126948666715, -0.5669126948666715, 0.07275552637463709]
accelerations: [21.37, 21.37, 0.0]
effort: [0.0]
time_from_start: 
  secs: 0
  nsecs: 106113747
Count :  3
POINT: 
 positions: [0.1818362839746781, 0.1818362839746781, 0.010568969776197397]
velocities: [0.0, 0.0, 0.0]
accelerations: [0.0, 0.0, 0.0]
effort: [0.0]
time_from_start: 
  secs: 0
  nsecs: 132642184
Count :  4
POINT: 
 positions: [0.17382536889095124, 0.17382536889095124, 0.017967653621135504]
velocities: [-0.5851380270316452, -0.5851380270316452, 0.42337658803633726]
accelerations: [-21.37, -21.37, 0.0]
effort: [0.0]
time_from_start: 
  secs: 0
  nsecs: 160023468
Count :  5
POINT: 
 positions: [0.14979262363977064, 0.14979262363977064, 0.02956024797841075]
velocities: [-1.1702760540632904, -1.1702760540632904, 0.42337658803633726]
accelerations: [21.37, 21.37, 0.0]
effort: [0.0]
time_from_start: 
  secs: 0
  nsecs: 187404751
Count :  6
[... 930 lines cut ...]
Count :  62
POINT: 
 positions: [0.27583195652995174, 0.27583195652995174, 0.0011662186765533911]
velocities: [0.0, 0.0, 0.0]
accelerations: [0.0, 0.0, 0.0]
effort: [1.0]
time_from_start: 
  secs: 2
  nsecs: 186345046
Count :  63
POINT: 
 positions: [0.27583195652995174, 0.27583195652995174, 0.0011662186765533911]
velocities: [0.0, 0.0, 0.0]
accelerations: [0.0, 0.0, 0.0]
effort: [0.0]
time_from_start: 
  secs: 2
  nsecs: 286345046
Count :  64
```
as I reduce the control cycle to 0.01 and lower the motion seems laggy during the return motion, i.e it seems like it is jumping steps or getting overridden by the next point. My objective is to achieve as low a control cycle as possible for smoother motions, We were earlier using UART and thought that was the bottleneck, but we have shifted currently to UDP and are sending msgs at 1.12ms times, so the current bottleneck now seems to be the way FastAccelStepper generates the motion and whether it is able to update at such high frequencies.

Your input would be greatly appreciated as you would have a better understanding of its capabilities, I am not able to figure out what the issue could be.

Thank you for your help in advance!

gin66, 2024-10-27:
You could try to reduce the forward planning time to a bare minimum using `setForwardPlanningTimeInMs(uint8_t ms)`. If the motor behaves strangely, then the queue has run out of commands. I would assume, that on esp32 5ms could still work. It must be for sure larger than 4ms stepper task cycle time. 

On  the other side, with this high degree of control of the motion is the ramp generator in use at all ?

gin66, 2024-10-27:
Another try is to reduce the stepper task cycle time below 4ms. This can be done only by changing the code in this function: https://github.com/gin66/FastAccelStepper/blob/e57f7b4fe676d5ea3ee44a0048b92d1770e5c7da/src/StepperISR_esp32.cpp#L107

If this is e.g. 1ms and forward planning time is e.g. 2ms, this should be quite performant. If the cpu can cope with this load is another question. I think you need to ensure, that wifi and stepper are running on different cores.

gin66, 2024-10-27:
If this is successful, then I can extend engine.init with cycle time as an optional parameter

programmeddeath1, 2024-10-27:
Hi Thank you for your quick response!

1) I have to set setForwardPlanningTimeInMs for each stepper right?.  ill try this and get back to you here.

2) Also I will incrementally test with 15 ms 10 ms and then go to 5ms. For now even reaching 8ms, would be a great improvement in performance for me.
The end goal being trying to push the device and see how performant it gets, I will later test in a more controlled environment, for going below 5ms, where changes need to be made in the ISR Code.

3) Regarding your question, I too had a doubt since you have mentioned the in the readme regarding multi-axis control that the  library's ramp generator would be less useful and its recommended to implement low-level access  directly using the library, At what point would the ramp generator prove insufficient? would that depend on how low I am going with my cycle time?

gin66, 2024-10-27:
Yes. The forward planning time needs to be set for each stepper and will be kept. In case a new stepper is allocated afterwards, then the forward planning time will be adjusted for all allocated steppers again. This will overwrite any earlier setting. If the forward planning time is set after all steppers are allocated, then the new value will be used and not overwritten anymore. This mean, if the steppers are allocated and configured in a loop, then the forward planning time need to be set in a second loop following the stepper allocation loop.

Thanks for your support in testing this functionality to the limits. Until now, I have only found a simulator for avr uCs. Great would be an esp32 simulator, so I can set up automated testing to check some of these advanced API features. Until then, those advanced features are only by design and not tested by me.

As I understand your use of FastAccelStepper the ramp generator will always ensure the limits for maximum acceleration and maximum speed. By design this is independent from the cycle time - unless there is any hardcoded value or implicit assumption in the code, which I do not believe, but am not 100% sure about. So there is no point, where the ramp generator will be insufficient. Just that the expectation of how the steppers should perform and the actual executed steps may not match, because of the constraints the ramp generator imposes on the stepper movements.

My expectation of the ramp generator is, that independent of the rate or magnitude of the changes in position/speed/acceleration, the stepper must always run within the limits of max speed/acceleration and always on the shortest round to the targeted position or speed.

programmeddeath1, 2024-10-28:
I tried adding ForwardPlanningTime as below - 
```
  engine.init();
  for (int num = 0; num < MOTOR_COUNT; num++) {
    // Input for Limit Pins
    pinMode(MOTOR_LIMIT_PINS[num], INPUT_PULLUP);
    stepper[num] = engine.stepperConnectToPin(MOTOR_PUL_PINS[num]);
    if (stepper[num]) {
      stepper[num]->setDirectionPin(MOTOR_DIR_PINS[num]);
      // You might also need to set enable pin if you have one
      setMotorSpeedInMMPerSec(num, SPEED_IN_MM_S);
      setMotorAcceleration(num, ACCEL_IN_MM_S);
    }
  }

  // Set future planning time limit for lower control cycles
  for (int num = 0; num < MOTOR_COUNT; num++) {
      stepper[num]->setForwardPlanningTimeInMs(10);
  }
````

I tried with values 5, 10, 15. I didn't see any difference in my motion. I will attach videos of my execution to help you understand what the issue could be:
Control Cycle = 0.02 secs
  
https://github.com/user-attachments/assets/c4eb0143-4654-4871-93a6-91e463d952ac

Control Cycle = 0.01 secs

https://github.com/user-attachments/assets/e4a2d040-e190-478e-9d35-9cd4be0da6cd

Control Cycle = 0.01 secs with fixed delay of 3ms

https://github.com/user-attachments/assets/a89207d0-1fc0-4413-aa99-1c7b8d61ae9a

The control Cycle of 0.01secs run with fixed dalay of 3ms, shows that the motion plan generates points based on smaller control cycle and  follows the actual C path, but the execution at the esp end is suspect.

gin66, 2024-10-28:
Thanks for sharing the video. Gives an idea, how you use the system.

Your sequence of the initialization is ok: First the motor, then the forward planning time

From the code in FastaccelStepper I only see, that in this line the condition should be put in brackets `()`
https://github.com/gin66/FastAccelStepper/blob/e57f7b4fe676d5ea3ee44a0048b92d1770e5c7da/src/FastAccelStepper.cpp#L386

Besides this, I cannot find anything suspicious in the code. If I find time the next days, I will make a test case running on simavr to check, that this function works as expected.

From the video I am not convinced, that the issue is related to FastAccelStepper. It could be network performance via UDP.  This connection may lag due to WiFi communication, esp32 UDP/wifi stack implementation or esp32 task management. I assume, there is no flash write somewhere in the code. Perhaps it is a good idea to add a timestamp to the UDP commands and check on esp32 side, if expected UDP commands have arrived too late.

gin66, 2024-10-28:
Thanks for checking, that UDP is not the issue. So there are two possibilities:
* Please modify the mentioned line and add the brackets `()` around the condition
* Skip the ramp generator and generate the commands directly from the UDP commands.

As mentioned, that I can add a test case only within a few days to check the general function. Besides the mentioned brackets, which may be the root cause, I am confident, that the feature is working on FastAccelStepper side.

programmeddeath1, 2024-10-28:
1) Sure, Ill make the changes and test once with the brackets around the condition.

2) Regarding skipping the ramp generator and generating the commands directly from the UDP commands, can you provide any references as to how to go about this? I checked your low-level access comments and got a gist of it, but some reference would be really helpful.

If we can identify that FastAccelStepper is not the issue, then we might have to investigate on the driver and motor end. Most AC Servo drivers come with response times in the range of 10us, so that was planned to be my last option to investigate. You can check on your end as and when you get time, ill make the changes and see if it yields me any results

gin66, 2024-10-31:
On master branch I have added a test case to check, if 
`setForwardPlanningTimeInMs()` works as expected and I have added the mentioned brackets for clarity. From my side, I do not see the reason for the observed behavior. Sorry for this.

From the video it appears, the stepper gets the command to stop or decelerate during movement. This does not hint to servo driver response time of physical limits of the motor.

In your first post, it can be observed, that the acceleration is in consecutive calls jump around the three values -21.37/0/+21.37. Could it be, that those initiate those stops ?

As this pick and place robot performs a complex motion for the steppers, it is not so obvious, what is correct behavior. How to determine, if the generated positions being sent via UDP are correct ?

programmeddeath1, 2024-11-02:
Hi the video named 1500RPM_motion_0.02cc_sliderview.mp4 , in one of the above commends runs at control cycle of 0.02, where it runs proper motion. The way we are verifying whether the positions are proper is I have run the entire trajectory generated at 0.01 control cycle, but added manual delays of 30ms between each point and run the motion as seen in  - 1500RPM_motion_0.01cc_3msdelay.mp4 -above. It shows the positions are proper.

How do we check if the accelerations/velocities are proper is currently only by running the motion physically :P sort of catch 22. But those acceleration spikes are normal, its following a trapezoidal trajectory generation, and the plots seem proper comparing to motions above control cycle of 0.02 which we could test physically. We do not currently have a physical simulator setup that can replicate the mass and inertia to simulate the acceleration and velocities properly, so that leaves us with testing on the actual hardware.

My current suspect is now the drivers, the datasheet does not have the exact stat so I am waiting on response from their team. A basic google search on the driver gave me the control frequency on these drivers as 50Hz but again not in any of their specification sheets. If it is 50Hz it could explain why we are being limited to 0.02 right now.

So once I can confirm that I can even test out using ForwardPlanningInMs with drivers at  higher control frequency say 100 or 200 Hz. I tested with the brackets, there seems to be no difference, so ill check on the drivers and get back.

gin66, 2024-11-02:
Thanks for explanation.What is still not clear to me, which of the stepper‘s parameters of position, speed and/or acceleration are controlled and how this is done? What could be of interest to monitor `rampState()`. 

The point is that for a given set of parameters, the ramp generator may decide, that stop needs to be executed and the next calculated set gives another result. If too high acceleration values the motion will be stuttering. 

If you control position and another value, then best sequence is to set speed/acceleration, then a move command. Do not call update `applySpeedAcceleration()`. It could be that between `applySpeedAcceleration()` and the move command, the stepper task is executed and this combination of new speed/acceleration and old position requires unwanted movements…just a thought .

programmeddeath1, 2024-11-02:
How I am executing the motion is as follows, 

```
void setMotorPosition(int num, float pos_mm) {
  if (stepper[num]) {
    float pos_steps = static_cast<float>(pos_mm * STEPS_PER_MM);
    stepper[num]->moveTo(pos_steps);
  }
}

void setMotorSpeedInMMPerSec(int num, float speed_mm_s) {
  if (stepper[num]) {
    float speed_steps_s = speed_mm_s * STEPS_PER_MM;
    stepper[num]->setSpeedInHz(speed_steps_s);
  }
}

void setMotorAcceleration(int num, float accel_mm_s2) {
  if (stepper[num]) {
    float accel_steps_s2 = accel_mm_s2 * STEPS_PER_MM;
    stepper[num]->setAcceleration(accel_steps_s2);
  }
}

void set_point(int num, float pos, float velo, float accel) {
  setMotorSpeedInMMPerSec(num, velo);
  setMotorAcceleration(num, accel);
  setMotorPosition(num, pos);
}
    for (int num = 0; num < MOTOR_COUNT; num++) {
      set_point(num, packet.positions[num] * 1000, packet.velocities[num] * 1000, packet.accelerations[num] * 1000);
    }

```
I set the velocity and acceleration, then run the moveTo for the next position. So any specific point in the path generated is basically the speed and acceleration needed right now to move to the next position in the required time. This ensures the end-effector reaches the object on the conveyor at the right time.

```
the stepper task is executed and this combination of new speed/acceleration and old position requires unwanted movements
```
This could be true for a  moment of time in microseconds i guess, but its running pick and place on a moving conveyor at control cycles more than 0.02. Of course, we have not tested on mm level precision, where this could actually be causing impacts coz of unwanted movements for that small a time.

gin66, 2024-11-02:
Thanks for sharing. 

In general I am always confused to see a float being fed into a function expecting an integer. Apparently it works, but I would expect the compiler to issue some warning. 

First, `setAcceleration()` expects only positive values and returns -1 in that case (even though the parameter is defined as signed integer). Second, I would assume, it has advantages to set the acceleration to the machine capability once and leave it unchanged. The movement is already fully controlled by position and speed. Similarly, `setSpeedIn..()` expects only positive values, too. A negative float will be converted into a big integer, which represents a pretty low speed. Any pulse representing low speed will be completed, even though new commands are coming in. The pause is not interruptible. If e.g. a pulse is generated with 1step/s and the step has just been generated, then the remainder of the time till 1s is complete, no other pulse will be generated.

Just my two cents from looking at the code. If the speed/acceleration values are all positive, then this can be safely ignored.

gin66, 2024-11-02:
The sequence of setting speed, acceleration and then position is completed fine and there is no possibility by task switching in between to cause an issue with unsynchronized updates. Only the `moveTo()` will apply the previously changed speed/acceleration.

vasekvanek, 2024-11-05:
Hello, I have a similar issue. At the beginning of the thread, it seems like a ROS trajectory output, which I also use. My question is about creating the command queue. If I understand correctly, the stepper creates a queue of commands that arrive faster than it can process each one, so it continues with the original target value and then moves on to the next in the queue – is that correct? If so, how is the queue generated over time? I'm trying to understand if, when I set speed and acceleration and then call moveTo, the motor moves... and if, before it finishes moving, another speed, acceleration, and moveTo are called, will the new speed and acceleration be applied only at the new position, or will they change immediately when moveTo is called?

gin66, 2024-11-06:
The ramp generator fills the stepper queue with commands and keeps track of position/pulse rate at queue end … normally some time in the future. For very low speeds e.g. 1step/s this future can be even 2 seconds: one command in process and one command prepared in queue. For a move to complete, it may need 1 or 10 or 100000 commands to complete - depending on the move. Those commands can be steps, multiple steps or pauses. The queue will be minimally filled to have 20ms planned in the future. Any new move/moveTo/stop/applySpeedAcceleration will only append command to queue end and will not change existing commands. If one move is 1 step at 1step/s, then the next command will be processed after completing the 1s.

gin66, 2024-11-06:
The ramp generator is processed every 4ms in cyclic interrupt or task depending on the uC

vasekvanek, 2024-11-06:
So the queue only stores the position?

gin66, 2024-11-06:
The struct is here:
https://github.com/gin66/FastAccelStepper/blob/f83a711c012b7e7557407acbc06cfd98fc0c5103/src/fas_arch/common.h#L50

gin66, 2024-11-06:
The ramp generator keeps its own queue end related data, which is defined here:
https://github.com/gin66/FastAccelStepper/blob/f83a711c012b7e7557407acbc06cfd98fc0c5103/src/RampConstAcceleration.h#L31

gin66, 2024-11-06:
The queue does not care about the current speed/acceleration, so the queue end structure only keeps position related info. The ramp generator need to know those values, so more data is kept.

If commands are removed from the queue, then the ramp generator’s data structure would be out of sync.

vasekvanek, 2024-11-06:
This means that for precise synchronized movement of multiple motors to achieve linear motion of my manipulator's end effector, the high-level application of FAS is not very suitable. It is not possible to accurately set acceleration and speed at transition states over time because, when I want to use Moveto at exact time intervals designated for movement, the previous motion is already decelerating at that moment? I tried to address this issue with a slight "lead," setting the commands earlier than the end of the movement, but this causes significant desynchronization.

vasekvanek, 2024-11-07:
Alright, I'll try to dive into it and see what can be done. Thanks a lot for the consultation. I really appreciate your work, as it allows me to do things I could hardly manage on my own. Do you have any suggestions on where to start studying the problem so that it makes sense for my project? And is there any description on how to work with the low-level aspects beyond the documented API? It's quite a complex topic for me... and I would need something detailed.

gin66, 2024-11-07:
some doc is in #86 

examples:
examples/Pulses/Pulses.ino
examples/RawAccess
examples/RawAccessWithPause

ChrGri, 2024-11-10:
Hi @gin66, I'm very much interested in high-frequency positioning. My ESPs control loop runs at 1ms or lower and likewise, the moveTo() is called. The servo I use has a step/dir interface that accepts pulses up to 300kHz. 

Any chance we get a simple API like
moveToPositionWithoutRamp(targetPosition), ignoring any acceleration limits and only taking into account the maximum pulse frequency of the interface? Repetitive calls of the function should overwrite the previous target position immediately. 

BR

vasekvanek, 2024-11-12:
I have another question: How is it possible to achieve high frequencies using "addQueueEntry"? According to the description, it is possible to have MIN_CMD_TICKS set to 3200, which translates to 5000 steps per second. Or did I misunderstand something?

gin66, 2024-11-12:
The minimal _command_ duration is defined by `MIN _CMD_TICKS`. If this value is 3200 ticks, then the command can be e.g. 1 step or pause with >=3200 ticks, or 10 steps at >= 320 ticks, or 20 steps at >= 160ticks,… A command can be either a pause or 1…255 steps with the related ticks.

vasekvanek, 2024-11-13:
Thanks, that's clear. Now I "just" need to calculate it so that the accelerations and decelerations work for my application
. So, is acceleration and deceleration a combination of incrementing ticks/steps in a single command?

gin66, 2024-11-13:
a single command has a fixed pulse rate. Acceleration and deceleration needs several commands in sequence.

vasekvanek, 2024-11-13:
That's what I meant, for example, 6000 ticks/10 steps, then the next command 5000 ticks/10 steps, and so on... acceleration and conversely deceleration, or the same number of ticks but with a decreasing or increasing number of steps. Is that correct?

gin66, 2024-11-13:
The algorithm is bit complex. For slow speeds (lower than 1kHz) normally each command contains one step. If the step rate is even lower than 65535 ticks, then pause commands will be included between steps. For higher speeds, the algorithm tries to settle on a 2ms command time by requesting several steps per command. Depending on the position on the ramp, or if during coasting the deceleration start is approaching, the number of steps and thus the command time can be decreased.

vasekvanek, 2024-11-13:
I understand that the implementation will be complex, but essentially it's about combining the pause between commands, tick and step, and tick/step (it’s important not to end up with a remainder or exceed the MCU frequency). I'll try to put it together somehow
It would be a great feature to simply input points with their attributes into a queue and create a path through multiple points, not just from A to B, which is insufficient for my application. If it works out, I’ll be happy to share the code. We could even add it here if needed ;)

ChrGri, 2024-11-20:
@gin66 I've implemented a very specialized and rudimentary version of this awesome library without acceleration and learnt a ton from your work. 

To compare the libraries, a [benchmark](https://github.com/ChrGri/FastNonAccelStepper/tree/main/examples/benchmark_1) has been setup, in which a sine wave target signal (10Hz) has to be followed. Examples of the pulse following are depicted below. The blue line is the target position, the green line is the current position from "FastAccelStepper" and the red curve is the current position from "Fast**Non**AccelStepper". 

"FastAccelStepper" lib:
[image]

"Fast**Non**AccelStepper" lib:
[image]

The FastAccelStepper settings from the benchmark script resulted in significant position lag. Furthermore, the "FastAccelStepper" trajectory looks a bit jittery. 

Do you have some settings in mind, which would improve the benchmark behavior?

BR
Chris

gin66, 2024-11-20:
Thanks for the new measurements. Actually I am surprised, that forward planning time 1ms has worked without motor stalls. Reducing the task period together with short planning time should improve the results.

gin66, 2024-11-20:
Thanks again for measurements. For esp32 adding the option to adjust the stepper task rate makes apparently sense. Then the app developers have more control about FastAccelStepper performance.

The 4ms task rate was just a carry over from avr. In order to have as much code overlap as possible and reuse the test suite from avr, the 4ms will remain to be the default. 

I do not remember to have hit a limit with esp32 load or bad performance, but the mcpwm/pcnt combo is sensitive to too long interruption blockages design. There is code included to recover from an overrun, but difficult to test. esp32 cpu load measurements haven’t been performed., but I know, that writing to flash while running a stepper is not a good idea. The stepper task will stop filling the queue, which leads to an abrupt stop and with delay a sudden restart at previous speed. At least no step will be lost, because the interrupts are in IRAM. For espidf 5.x I am not so sure anymore, because here a good deal of espressif code is in use.

programmeddeath1, 2024-12-24:
> g
In our case, the negative values simply signify the direction of the motion in the case of speed. So we have negative acceleration which will increase the speed in the negative direction. Positive acceleration will increase the speed in the positive direction of motion. As can be seen from the first few values of position, speed and acceleration commands in the first comment.  In case of float data being sent, i believe the compiler simply truncates the float to integer and hence the function runs smoothly.

Although If negative accelerationis not accepted by the function call setAcceleration, do you mean to say that the motion is running purely based on the position and speed controls provided by the trajectory? My question is for control cycles more than 0.02, the motion is quite accurate, and we seem to be facing stops or speed change issues only at higher update rates (< 0.02), should I try running the motion with signed int values for speed to see if the issue still persists below 0.02 control cycle. 

For me the current culprit is the driver which we are in the process of upgrading. Based on the above plots by users above , it seems the forwardPlanning is working even for upto 1ms, so can we for sure claim this is not an issue with the library?

gin66, 2024-12-25:
„For sure“ is difficult to say. It highly depends on the other code running on the esp32 in which FastAccelStepper is a part of. The OS is normally configured with 1ms task rate. So 1ms delay in the StepperTask loop may result in the StepperTask only running every 2ms.

programmeddeath1, 2024-12-25:
Yeah but still my current requirements are not that high either. I am working with 20ms currently, so even if I can get the cycle down to 10ms or 5ms, we are still reaching industrial motion controller standards. Upto 10ms I guess should be feasible or do I need to run cpu utilization scripts parallel with the stepper calls to confirm if this works? I'm currently stuck between both where I am unable to confirm if the bottleneck is the driver or the esp32.

gin66, 2024-12-25:
@ChrGri has obtained good results at 2ms delay. Perhaps you can try something like 2ms delay and 5ms forward planning

closed 2025-02-20 by gin66