gin66/FastAccelStepper #289: Support for smooth motion

tomassasovsky, 2024-10-28:
Hey there! I'm trying to make simple movements with the motor, having it just move at a constant speed. I'm facing an issue where the motor makes a stuttering sound, sorta sounds like a locomotive. The motion is not smooth at all, it's very interrupted.

I'm using the code from the example:

```
#include "FastAccelStepper.h"

// As in StepperDemo for Motor 1 on ESP32
#define dirPinStepper 18
#define enablePinStepper 26
#define stepPinStepper 17

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  if (stepper) {
    stepper->setDirectionPin(dirPinStepper);
    stepper->setEnablePin(enablePinStepper);
    stepper->setAutoEnable(true);

    // If auto enable/disable need delays, just add (one or both):
    // stepper->setDelayToEnable(50);
    // stepper->setDelayToDisable(1000);

    stepper->setSpeedInUs(1000);  // the parameter is us/step !!!
    stepper->setAcceleration(100);
    stepper->move(1000);
  }
}

void loop() {}
```

gin66, 2024-10-28:
I would assume, this is not a sw issue. Could you please check the cable ? Especially ground connections are often cause of unexpected behaviors.

tomassasovsky, 2024-10-28:
Very well might be. Do I need to add any capacitors, resistors, etc. to my circuit?

gin66, 2024-10-28:
I do not know your set up. Do you have an oscilloscope or similar to check, that the steps are generated properly ? Or just start without FastAccelStepper and control the stepper with a loop, which contains just a toggle of step pin and a delay ?

tomassasovsky, 2024-10-28:
I don't have an oscilloscope, no. Could you provide an example snippet for that?

gin66, 2024-10-28:
Something like:

```
// As in StepperDemo for Motor 1 on ESP32
#define dirPinStepper 18
#define enablePinStepper 26
#define stepPinStepper 14

void setup() {
   pinMode(dirPinStepper, OUTPUT);
   pinMode(stepPinStepper, OUTPUT);
   pinMode(enablePinStepper, OUTPUT);

   digitalWrite(dirPinStepper, LOW);
   digitalWrite(stepPinStepper, LOW);
   digitalWrite(enablePinStepper, LOW);
}

void loop() {
   digitalWrite(stepPinStepper, digitalRead(stepPinStepper) == LOW ? HIGH:LOW);
   delay(50); // => one step in 2*50ms, so 10 steps/s
}
```

tomassasovsky, 2024-10-28:
I tried your code with a delay of 1ms  which is closer to the speed I need, and it's very noisy.

https://github.com/user-attachments/assets/1ffbf29e-eb48-4f73-a4be-34a9379bf9a6

gin66, 2024-10-28:
Thanks for sharing the video. If this low level code does not work, then something is wrong with the hardware.: Ground connection, assignment of dir/step/enable to the gpio, cable from stepper to stepper controller - especially the pin assignment, is the power supply strong enough, are the cables not too long, is the uC constantly rebooting….? Good luck in your endeavor

gin66, 2025-02-20:
stale