gin66/FastAccelStepper #290: ESP32 Not Generating Step Signal

vepicurean, 2024-11-04:
I have been using FastAccelStepper for years now. Never ran into this before. 

I set up a new ESP32 ( expressif ESP32-WROOM-32E) and I just can't get a signal from the assigned step pin. No matter what I do. I even used your prepared demo, no go. 

I even went back and set FastAccelStepper to version 0.23 and the ESP32 board library version to 1.0.6 like I've successfully used in the past via the Arduino IDE. No go. The stepping signal is dormant ( like it failed for some reason). I measured the step pin with a voltmeter. It's not floating, but the step PIN measures 0VDC, no signal. 

The output voltages on the direction and enable are fine for the high voltage for the step controller to register. 

I get the direction signal output and the assigned enable when using fastaccelstepper, so that functionality works. No step signal. 

When I simply put a high or low on the GPIO pin assigned to the stepper ( not using FastAccelStepper) the signal on that is registered by the step motor controller with no problems. 

I'm using the latest version of the Arduino IDE ( had to do a few things in the background to get the Python 2 compiler reference to point to Python3 - but I got that to work). It compiles at least with no errors. 

I tried to debug it for two days. Can't figure it out. 

Any suggestions?

gin66, 2024-11-04:
Perhaps same root cause as in #254 ?

vepicurean, 2024-11-05:
> Perhaps same root cause as in #254 ?

Thanks for the reply. 

If I understand #254 correctly, hopefully this is something new specific to these Arduino Nano ESP32 boards, as I've never run into this before on my ESP32 WROOM-32E boards, for the step pin only, I have to use the physical pin number, not the GPIO number? 

I've always used the GPIO number on these expressif ESP32 boards ( WROOM-32E) and never had an issue getting FastAccelSteeper library to output steps. 

[image]

Is this a new thing? I just want to be sure before I start changing assignments on my long standing design.... I'm not using an Arduino Nano ESP32, I know those Arduino Nanos are different ESP32 boards to the expressif boards. 

Thanks as always.

gin66, 2024-11-05:
Thanks for the update. You refer directly to the espressif module and not an arduino board. In this case, there should not be a mismatch of pin versus gpio number. So there is no obvious reason for the behavior.

What you can try:

1. select the rmt driver on stepperConnectToPin()
2. issue #245 mentions, that the init function do not evaluate return codes during module init. Are there any warnings/errors during execution of connectToStepperPin() ?
3. If dir pin is properly toggling based on move commands and the mcpwm/pnct- driver is used, then there must be pulses generated. Without the feedback loop working, the very first move commands will never end, but stall at the first step. Is any other pin toggling ?

vepicurean, 2024-11-06:
> Thanks for the update. You refer directly to the espressif module and not an arduino board. In this case, there should not be a mismatch of pin versus gpio number. So there is no obvious reason for the behavior.
> 
> What you can try:
> 
> 1. select the rmt driver on stepperConnectToPin()
> 2. issue [ESP32-S3, RMT driver on 5th stepper fails #245](https://github.com/gin66/FastAccelStepper/issues/245) mentions, that the init function do not evaluate return codes during module init. Are there any warnings/errors during execution of connectToStepperPin() ?
> 3. If dir pin is properly toggling based on move commands and the mcpwm/pnct- driver is used, then there must be pulses generated. Without the feedback loop working, the very first move commands will never end, but stall at the first step. Is any other pin toggling ?

Thanks for the reply. 

I looked all over, but I stuggling to understand what you mean be #1. Select the rmt driver on stepperConnecttoPin() 

The step, dir, and enable outputs from the ESP32 board simply interface two the inputs on the controller for the step motor. The controller reads the signals for the outputs if the ESP32 actually sends them. I have older setups between an ESP32-WROOOM-32E and the same controller. They work just fine. 

There are now warnings or errors during the setup. Just no signal. The ESP32 thinks is generating steps, but it's not. The ESP32 chip does not even get warm.  Usually, when it's generating steps it gets warm to the touch. 

The dir and enable pins work just fine. It's the STEP pin output that does not work.

gin66, 2024-11-06:
I mean: `engine.stepperConnectToPin(stepPinStepper, DRIVER_RMT);`

gin66, 2024-11-06:
You can attach a pulse counter to the step pin using:
```
  stepper->attachToPulseCounter(QUEUES_MCPWM_PCNT, 0, 0);
  stepper->clearPulseCounter();
```
and monitor the value while the stepper is running using `int16_t pc = stepper->readPulseCounter()`. If this value is not changing, there are no pulses generated.

vepicurean, 2024-11-06:
> I mean: `engine.stepperConnectToPin(stepPinStepper, DRIVER_RMT);`

I've never had to use this DRIVER_RMT before. Is this a recent feature addition? I'm using 0.23 of FastAccelStepper

I get a comilation error about DRIVER_RMT not defined when I try to use in with engine.stepperConnectToPin(stepPinStepper, DRIVER_RMT);

gin66, 2024-11-06:
Please upgrade to lastest FastAccelStepper version 0.31.2

vepicurean, 2024-11-06:
I tried that at the start ( I had everything at the latest version), but I was having trouble even then with generating the step signal. 

So I figure let's go to the same board version and fastaccelsteeper version for this ESP32-WROOM-32E that I've used for years and know works, but so far that's not solving the issue with this WROOM-32E ( which is one of the first generation ESP32's I believe).

vepicurean, 2024-11-06:
> So nothing has changed, but it has stopped working on the new boards ? I assume, then there is no issue with FastAccelStepper. Hope you will find the problem. Perhaps a hw issue ? Classical problem is an issue with ground connection.

No, it was getting the wdt error with the latest version. fast fastaccelstepper was the cause of the wdt error. I first had to get rid of the wdt error, by going to the version that I know worked before, but the step signal does not work.

vepicurean, 2024-11-06:
> If you search for wdt in the issues, you may encounter #283 and there is the working solution to use platformio. And again, if you go back to the old sw version and that has worked before but not anymore, then FastAccelStepper may not be the root cause of the problem and eventually there is a hw issue ?
> 
> BTW: in order to diagnose a problem, it would be helpful, that you try some of my proposals and provide feedback like log-files from Serial. Otherwise I cannot help you. And repeating, that the new version does not work and the previous working old versions do not work anymore, does not give sufficient clue - at least to me. And there is no proof, that your hw is without flaws. That a FastAccelStepper example does not work, tells me nothing, because I do not know, if the signals have been updated to your hardware. And again, a log file may give clues about any underlying problem. And if not, then still is more helpful than a “does not work”.
> 
> As I am soon out for vacation, I may not find time the next eight days to provide feedback.

I understand. I worked on it for three days, doing as you are mentioning before contacting you. 

Yes, I looked through the forum and other sources for days trying to find a solution to the wdt issue. Those workarounds did not work for me. Appreciate your suggestions. I did much as you suggested already. 

On the test demo, I get this compilation error on multiple WROOM 32Es. I looked through the forum and not seeing anything to address this error using the DRIVER_RMT paramenter.   The fastaccelstepper is at the latest version 0.3.15 for this demo. 

error: 'DRIVER_RMT' was not declared in this scope
    stepper = engine.stepperConnectToPin(stepPinStepper, DRIVER_RMT);

gin66, 2024-11-06:
0.3.15 does not exist. 0.3.14 is fromJune 2020. latest is 0.31.2

`DRIVER_RMT` is introduced in 0.29.2 for esp32 variants, which support mcpwm/pcnt and rmt. Apparently, ESP32-WROOM-32E is a regular esp32 and fits. mcpwm/pcnt-driver is not available anymore for espidf v5.x, but espressif-arduino is still not with espidf 5.3. Only the espressif-arduino rc2 is: https://github.com/espressif/arduino-esp32/releases

vepicurean, 2024-11-06:
I seem to have figured it out for the test demo code. 

I going to spend some time repeatong this other ESP32-WROOM-32E dev boards to make sure the solution is repeatable with generating the steps. 

The key was I had to go all the way back to Arduino IDE 1.8.13 - That previous version of the Arduino IDE combined with using the expressif board version 1.0.6 and the FactAccelStepper 0.23.0 works. 

Your suggestion to try a different IDE tipped me off that the IDE environment makes a difference. I figured this simple demo/test should be working without having to change the IDE, so I decided to go back to the IDE where I know it has worked in the past. 

Thanks for your help. Appreciated. I'll let you know how it goes.

closed 2025-02-20 by gin66