gin66/FastAccelStepper #292: (suggestion) Independend setting of "setDelayToEnable" and "setDelayToDisable" for the enablePins

Ramadama-san, 2024-11-11:
Hello,

for my project need to use a motor brake to avoid  movement of the motor if stopped.
Unfortunately, I cannot keep the motor enabled for an extended time.

So I'd like to use the two enable pins for the motor and the brake with a sequence where I can define the delay
of both pins separately.

Examples for
Start sequence:
1) enable Motor  
2) disable motor break (after delay of "setDelayToEnable(enablePin1, xx ms)"  )
3) start stepping           (after delay of "setDelayToEnable(enablePin2, yy ms)"  )

Stop sequence:
3) stop stepping
2) disable motor break (after delay of "setDelayToDisble(enablePin2, aa ms)"  )
1) disable Motor  (after delay of "setDelayToDisble(enablePin1, bb ms)"  )

I'm willing to try to modify the code but I am quite a newbie to coding on this level so I'm not sure if I will succeed :-(
I appreciate if you can help me, at least with a hint where to find the respective code.

Thanks

gin66, 2024-11-11:
Thanks for the proposal. If I understand correctly, you need two enable signals instead of one. 

The internal implementation of shared internal pins is not really great and a generic rework would make sense. The use case of two independent enable signals is rather application specific and IMHO better implemented on application side. A possible strategy is to use the external pin feature. The callback is invoked on each pin change repeatedly every 4ms until the callback returns the requested value. In the case of two pins, the callback would return the old pin value until the appropriate timeout is met.

Sequence could be:
1. Motor shall start, so callback is called with “enable”
2. Callback resets counter and sets motor break. Returns “disable”
3. Callback is called every 4ms and counter gets incremented.
4. Callback has met the counter value for delay to enable. So the motor is enabled and the callback returns “enable”
5. FastAccelStepper waits the configured delay time and starts stepping.

Disable is similar, just that FastAccelStepper calls “disable” after the delay is completed. The callback need then disable motor and after counter reaching the required delay value enable the motor break.

Ramadama-san, 2024-11-12:
Thanks for the quick reply.

I'll try to implement as you've proposed.
As said, I'm a newbie to coding on this level so it may take some time....

closed 2024-11-12 by Ramadama-san