gin66/FastAccelStepper #294: Add ability to define RMT channel

gmint, 2024-12-02:
Would you consider adding the ability to specifically define the RMT channel used on ESP32? 

My use case is as follows: I am using FastLED in conjunction with FastAccelStepper and although I should have enough RMT channels (i.e. four) on my ESP32-C6 to drive a small number of LEDs and one stepper motor, I am experiencing a condition where FastAccelStepper is unable to find sufficient TX channels even when I restrict FastLED to one RMT channel. Here is the error I am receiving:

16:01:41.268 -> E (4980) rmt: rmt_tx_register_to_group(152): no free tx channels
16:01:41.268 -> E (4980) rmt: rmt_new_tx_channel(288): register channel failed

I am currently able to workaround this by switching to LEDC but given the robustness of this library, it seems like something that might not be out of the question. Thank you for your consideration.

gin66, 2024-12-04:
Thanks for the proposal and your compliment about the robustness of FastAccelStepper.

Good news is, that with espidf 5.x the rmt channel assignment is managed by espressif and FastAccelStepper supports >=5.3

espidf5 restricts direct access to the hw module and control of the channel is not possible. This means the proposed change would be limited to espidf4 versions. This older rmt driver hijacks the module interrupt, because there is no Interrupt service handler. This means cooperative sharing the rmt module with other libs is complex. In addition, there are three versions of rmt-drivers for the different esp32 derivates. As a result, I do not see this as a worthwhile endeavor.

Just I am puzzled about the error message with `rmt_new_tx_channel(288)`. If this is caused by FastAccelStepper, then this would indicate, you are using the espidf5 versions. Hope you can further elaborate on the versions in use.

gmint, 2024-12-04:
That’s interesting because I am, in fact, using the RC of the 5.3 branch of the ESP-IDF for Arduino. I’ve tried the config variable in FastLED that is supposed to limit the number of RMT channels that it uses (which, if my understanding is correct, is all of them if it can). If there’s anything I can do to further elaborate, please let me know!

gin66, 2024-12-05:
Do you configure mem_block_symbols!=0 in led_strip_rmt_config_t of  FastLED ? Perhaps FastLED uses all rmt memory ?

gin66, 2024-12-05:
https://github.com/FastLED/FastLED/issues/1768  related ?

gmint, 2024-12-05:
I didn’t but I will try. Also to be clear, either FasrLED or FastAccelStepper compile and work fine on their own. It’s just the combination that errors out.

gmint, 2024-12-05:
I’m not sure if this is or not. I did report an error with the ESP32-C6 that was fixed in a pre release version of FastLED.

FastLED/FastLED#1768  related ?

gin66, 2025-01-31:
stale

closed 2025-01-31 by gin66