gin66/FastAccelStepper #295: Cyclic reboot ESP32 Dev Module

iSetebos, 2024-12-05:
Cyclic reboot when switching to engine.init(); . Arduino 2.3.4. Library 0.30.15.

ELF file SHA256: c69cac15c
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  1 register dump:
PC      : 0x4008453c  PS      : 0x00000016  A0      : 0x40080306  A1      : 0x3ffc3470  
A2      : 0x3ffb1efc  A3      : 0x00000000  A4      : 0x00100590  A5      : 0x3ffc34a0  
A6      : 0x00000000  A7      : 0x3ffc3030  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000020  A12     : 0x3ffb1e80  A13     : 0x70000005  
A14     : 0xfffbfff0  A15     : 0x3ffc293c  SAR     : 0x00000008  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40085e60  LEND    : 0x40085e76  LCOUNT  : 0xffffffff  

Backtrace: 0x40084539:0x3ffc3470 0x40080303:0x3ffc34a0 0x65a5a5a2:0xa5a5a5a5 |<-CORRUPTED
ELF file SHA256: c69cac15c
Re-entered core dump! Exception happened during core dump!
Rebooting...
ets Jul 29 2019 12:21:46

gin66, 2024-12-05:
Thanks for the report. Do you have some source code, which reproduces the issue ?

iSetebos, 2024-12-06:
Example UsageExample.ino

gin66, 2024-12-06:
Thanks for sharing. The [`init()`](https://github.com/gin66/FastAccelStepper/blob/0585f4661db78d0f3ebd78f676ae9f4f96e3bc35/src/FastAccelStepper.cpp#L21) does only perform few variable initialization. Does any sketch without FastAccelStepper work ?

iSetebos, 2024-12-06:
Yes. Many projects have been written.

iSetebos, 2024-12-06:
The problem here is

[image]

gin66, 2024-12-06:
Thanks for pin pointing this down and I have overlooked that task creation is part of `engine.init()`

Anyway, this is another good example for one of nearly all Guru meditations, that the provided information is no help for debugging.

The `xTaskCreate()` is commonly used since years in FastAccelStepper and is expected to create a task successfully. That it fails in your setup is strange. Which esp32 do you use ?

iSetebos, 2024-12-07:
ESP32-WROOM-32

gin66, 2024-12-07:
Thanks for the info. This is just the base mode and should work. Please try platformio. This is more professional, less [buggy](https://github.com/arduino/libraries-repository-engine/issues/194) and the IDE I use.

iSetebos, 2024-12-07:
I reinstalled everything and now the following error.
E (21) rmt: rmt_new_tx_channel(216): mem_block_symbols must be even and at least 64
Error creation 258

gin66, 2024-12-07:
Thanks for your patience and reinstalling has made progress. Now `engine.init()` does not fail. Pity, that it is not clear, what the root cause was.

The arduino library manager is buggy and 0.31.3 is not available. Which lib version of FastAccelStepper is now used in the reinstalled setup ?

iSetebos, 2024-12-07:
0.30.15

gin66, 2024-12-07:
please try to manually upgrade to 0.31.3.

iSetebos, 2024-12-07:
Thank you. It works!!!

closed 2024-12-07 by iSetebos