gin66/FastAccelStepper #296: motor not working when using setAbsoluteSpeedLimit

janajoe, 2024-12-10:
I use 3 stepper motors (class mymotor) which I initialize like this:
    stepper = engine.stepperConnectToPin(pins.pulPin);
  
    if (stepper)
    {
      stepper->setDirectionPin(pins.dirPin);
      stepper->setAutoEnable(true);
      stepper->setSpeedInHz(speedStart * microsteps);
      stepper->setAcceleration(accelStart * microsteps);
      // stepper->setAbsoluteSpeedLimit(vMaxGlobal * microsteps);
    }
They work fine when I don't use setAbsoluteSpeedLimit but using setAbsoluteSpeedLimit, only two work fine and the third one doesn't move at all. It seems to get weird signals over the pulse pin. It's always the third motor that I initialize that doesn't work.

gin66, 2024-12-10:
Which uC do you use ? In addition, please output the speed limit:
```
uint16_t limit = vMaxGlobal * microsteps);
Serial.print("limit=„);
Serial.println(limit);
stepper->setAbsoluteSpeedLimit(limit);
```

janajoe, 2024-12-10:
I am using an ESP32 (Sparkfun thing plus).
The limit is output as limit=62464 for each of the motors.

I also noticed, that for the first motor that is initialized, I get "E (17126) pcnt: _pcnt_isr_service_install(317): ISR service already installed" just after using stepperConnectToPin...But it is still the setting of the absolut speed limit row that leads to all 3 motors working as expected or not..

gin66, 2024-12-10:
Thanks for clarification.

The documentation states: **Allocating a new stepper will override any absolute speed limit.**

After allocating the third stepper, the maximum speed of the first two has been overwritten to default values. This means, only the third stepper is using the limit. The purpose of this call is to increase the speed limit beyond the chosen safe limits. Using 62464 as value means, the max speed is 256 steps/s for one command. But it could be, that the ramp generator creates at that speed two commands with e.g. pause of 31232 and a single step with 31232 and both will be rejected. If you need really max speed of 256 steps/s, then do not use setAbsoluteSpeedLimit() and limit the speed on application side.

janajoe, 2024-12-10:
ok thank you very much!

closed 2024-12-10 by gin66