gin66/FastAccelStepper #297: I get the error "task not found".

shin-man, 2024-12-13:
I'm currently using the ESP32-WROOM-32E to start off with a simple continuous rotation, but the following message is continually displayed on the serial monitor, and the motor shows no signs of moving at all.

「task_wdt: esp_task_wdt_reset(763): task not found」

What's wrong?

shin-man, 2024-12-13:
By the way, even when I tried the code from issue #289, the situation did not change.

shin-man, 2024-12-13:
I have seen the past issues.
Will using ”FastAccelStepper v0.31.2” solve the problem?

gin66, 2024-12-13:
You can give it a try.

shin-man, 2024-12-17:
As this was my first time using PlatformIO, it took some time, but when I used "FastAccelStepper v0.31.2" with PlatformIO, I was able to turn the motor.
Is there a problem with the Arduino IDE that the latest version of the FastAccelStepper library does not appear in the Arduino IDE?

gin66, 2024-12-17:
As written at the top of the readme:

# BE AWARE: ARDUINO LIBRARY MANAGER IS BROKEN AND 0.29.x/0.30.x with x>0 do not show.

No issue with platformio. Check the [related issue](https://github.com/arduino/library-registry/issues/2829) for the arduino library manager

shin-man, 2024-12-17:
I've just started using microcomputer control, so I didn't realize it and didn't know how to use it.
I'm sorry.

closed 2024-12-17 by shin-man