gin66/FastAccelStepper #298: Both the runForward and runBackward functions rotate in the same direction.

shin-man, 2024-12-17:
Since I was able to rotate it using the runForward function, I tried to use the runBackward function to rotate it in the opposite direction, but it ends up rotating in the same direction.
In a past question, someone was using two motors and had a similar problem, but in my case I'm only using one motor.
When I physically disconnected the wiring to the motor driver from the pin assigned to the Step signal, it started rotating in the opposite direction.
The microcontroller I'm using is an ESP32-WROOM-32E.
What's wrong?

shin-man, 2024-12-17:
私のコードを記載しておきます。

#include <Arduino.h>
#include <FastAccelStepper.h>

// Pin Settings
#define STEP_PIN 26  // STEP pin（Connect to GPIO 26 on the ESP32）
#define DIR_PIN  27  // DIR pin（Connect to GPIO 27 on the ESP32）

// An instance of the FastAccelStepper object.
FastAccelStepperEngine engine;
FastAccelStepper *stepper = nullptr;

void setup() {
  // Start serial communication
  Serial.begin(115200);
  while (!Serial);

  // Initialize the FastAccelStepper engine
  engine.init();

  // Stepper motor settings
  stepper = engine.stepperConnectToPin(STEP_PIN);
  if (stepper) {
    // Set directional pin
    stepper->setDirectionPin(DIR_PIN);

    // Set maximum speed and acceleration
    stepper->setSpeedInHz(1000);    // Maximum Speed（Hz）
    stepper->setAcceleration(6400); // Acceleration（step/s^2）
  } else {
    Serial.println("Stepper connection failed!");
  }
}

void loop() {
  digitalWrite(DIR_PIN, LOW);
  stepper->runForward();   // Rotate in the forward direction
  delay(5000);             // Wait 5 seconds

  stepper->stopMove();     // Stop
  delay(1000);             // Wait 1 seconds

  digitalWrite(DIR_PIN, HIGH);
  stepper->runBackward();  // Rotate in reverse

  delay(5000);             // Wait 5 seconds

  stepper->stopMove();     // Stop
  delay(1000);             // Wait 1 seconds
}

gin66, 2024-12-17:
Sounds like a connection issue. Good start is to check the ground connection between uC and stepper motor driver.

closed 2024-12-17 by shin-man