gin66/FastAccelStepper #299: Bypass ramp generator

AntoineGR, 2024-12-18:
Hello Gin66,
would you consider implementing a native command to run moves without ramp generator ?

The idea came up in previous use cases (#110 #288 #86) where motion planning is computed by an external software (3D, animation, robotic).
FastAccelStepper is then used to run many small consecutive motions (from an array in memory or streamed on the fly).
This allow for clean and time-constrained motion paths to be executed.

To bypass the current ramp generator some users tried setting the acceleration to max value (not 100% precise and use CPU ressources) or implemented their own low-level routine and queue management.

So the proposal would be to have two new commands that run in a given time without accel/decel :
`moveTimed(steps, duration)` and `moveToTimed(steps, duration)`
Duration could be in ticks or seconds or milliseconds.

gin66, 2024-12-19:
Thanks for the proposal. Sounds easy, but solution is a bit more complex.

Assume duration is 5ms and 50kSteps/s. Then steps = 250 and duration 5ms. Step rate is 20us. Easy. Can be done by simple translation into queue commands.

Now consider 251steps and duration 5ms. Step rate should be 19,92us. But the next lower step rate is 19.9375us and 251 steps with this rate is 5,004ms. So 4us deviation. With 252steps the actual duration is 4,99275ms and 8us deviation. Sounds small, but after 1000 commands of 5ms, there can be already 4-8ms deviation, which is a complete command. so ~250 steps more or less at nearly the same time.

Correction can be done only up to 4 times per ms and the deviation may not be possible to perfectly compensate. Shall this be corrected by the new function or by the caller ?

AntoineGR, 2025-01-09:
Thank you for the quick and thoughtful answer

You're making a good point on this quantization issue, I'm pondering what could be the most elegant solution,
will report back here soon !

AntoineGR, 2025-01-09:
Ok so in the spirit of my proposal the important point is to have _time-constrained_ commands.
Therefore the proposed functions must respect the desired duration and would sometime need to :
- accomplish less steps (to avoid overflowing the target duration)
- and wait at the end of the command (to reach the target duration)

So when using `moveTimed(steps, duration)` or `moveToTimed(steps, duration)` the functions should return `droppedSteps` to let the user know the delta and add it to the next function call.
Does that make sense ?

PS : I created a python script to loop through all stepping rates and check how many steps need to be dropped and how long is the remaining wait time, let me know if you'd like to have it.

gin66, 2025-01-10:
Thanks for the proposal.

Based on the existing implementation of the queue another approach seems to be more straightforward: The queue tracks very well the actual position at queue end. What is currently not available, is the time at queue end. This should be comparably easy to be maintained. Queue end time can be reset to 0 at start of the queue. Now the consecutive function calls would need to maintain there own set of target steps and target time. By comparing to the queue end time, the correction should happen quite naturally. 

Hope I have explained my idea good enough. What do you think?

AntoineGR, 2025-01-11:
Hi !
Sound alright but I'm not sure I understand all the implications : would the functions then return an absolute position + elapsed queue duration ? (maybe if you can give an example that could clear it out for me)

Continuously correcting for time might get tricky, especially with several motors working together.
I think the main use case of these function would be to run preplanned moves (or pathtables) entirely computed beforehand.

gin66, 2025-03-08:
Implementation is released in 0.31.5

gin66, 2025-12-26:
what do you mean with „losing steps“ ? Does the motor stops during the motion ?

gin66, 2025-12-27:
not sure, if the issue is resolved. so I reopen it to better keep track

Matoseb, 2025-12-29:
> Calling move stepper in every loop() invocation may not be, what you want. Perhaps include it inches if{} branch.

Ok this seem to work with no lost steps, but I can feel and hear the frames increments, you can see the steps in the curve:
Blue: target frame, Orange: current steps
[image]

---

Calling on each frame makes the movement smoother and with no lost steps, but there is a slope at the start and end of the movement, i would guess the commands are finished before the next frame:
[image]

---
Codes here [60fps-moveTimed.zip](https://github.com/user-attachments/files/24369409/60fps-moveTimed.zip)

My question is, how can I make it as smooth as your example movedTime while having frames fed at a constant 60fps rate?

gin66, 2025-12-29:
here are some comments to the frame version. Hope this helps:

```
void loop() {

  static elapsedMicros current_micros = UPDATE_INTERVAL_US;
  static elapsedMicros current_print = SERIAL_INTERVAL_US;

  if (current_micros >= UPDATE_INTERVAL_US) {
    moveStepper(stepper, &control);
    // adjust to the next trigger point.
    // need to point out: UPDATE_INTERVAL_US = 16.666
    //                    TIMESTEP_TICKS = 266.666 = 16*16.666 + 10
    //           for longer runs, the +10 may cause an issue.
    current_micros -= UPDATE_INTERVAL_US;
    control.phi += 1; // simulate next frame

    // output only, if moveStepper() has been just performed
    if (current_print >= SERIAL_INTERVAL_US) {
      // The Serial.prints are sending approx. 50Bytes, which needs 4 ms to transmit at 115.200 Baud
      // Let's do this only after next stepper command.
      int32_t current_position = stepper->getPositionAfterCommandsCompleted();
      Serial.print("Pos:");
      Serial.print(current_position);
      Serial.print(",Targ:");
      Serial.print(position);
      Serial.print(",Min:");
      Serial.print(-1600);
      Serial.print(",Max:");
      Serial.print(1600);
      Serial.print(",Phi:");
      Serial.print(controlX.phi);
      Serial.println();
      current_print -= SERIAL_INTERVAL_US;
    }
  }
```

gin66, 2025-12-29:
Thanks for further explanation about your set up and stripping down the code. Perhaps the next days I can convert it to a test, which I can run on my setup for reference.

Empty and Device Not Ready are indications, that the queue has or is in process to run empty. This means, the commands are not fed to the queue fast enough. (Device not Ready is a transient state, when the queue is in process of shutting down: not stopped, but not yet ready for new commands).

The two separate (nearly identical) time bases are apparently not doing the trick. Consequently, recommendation is  that moveTimed() is the only time base and ideally, the move commands for the next frame is provided in advance. This means instead of using current_micros as time base for the next move, only control.time is used. This should ensure, that always a full frame's move is in the buffer. As this may lead to e.g. Busy when the queue is full, on each loop iteration the moveTimed() may need to be repeated until OK is delivered.

Matoseb, 2025-12-30:
> Device not Ready is a transient state, when the queue is in process of shutting down: not stopped, but not yet ready for new commands

Got it

> Consequently, recommendation is that moveTimed() is the only time base and ideally.

Mmmh ok I see. So a solution for streaming positions in real time, would be to have a frame buffer that stores some frames in advance and then have control.time handle the frame seeking.

> Busy when the queue is full, on each loop iteration the moveTimed() may need to be repeated until OK is delivered.

So that's why the example has moveTimed called on each loop

gin66, 2025-12-30:
> So that's why the example has moveTimed called on each loop

yes. The example uses a simple approach: It appends moves to either stepperX or stepperY depending on the (future) time comparison until busy is coming. On busy, the loop will be only spinning (do nothing) until the queue has sufficient space for the next moveTimed. Drawback is that a full queue does not allow to react on fast changes. For better understanding of the control algorithm it is important to note, that the time decisions are on future time - after queued commands are completed, which may be several tenths of millisecond in the future.

For e.g. an XY-plotter the whole path of the complete design is predetermined and no changes need to be reacted on, so full queues are no problem at all.

gin66, 2026-09-12:
stale

closed 2026-09-12 by gin66