gin66/FastAccelStepper #307: pcnt unit 6 and 7

diarus, 2025-01-20:
Hello. Please help me. I'm need to use esp32 counter unit 7 in my project. Why do fastaccelstepper and counter work correctly only when using DRIVER_RT? Does the esp32 reboot in other modes? The description indicates that the 6th and 7th units of the counter are free.
There are 2 motors in the project.
When starting the  project, the code causes the esp32 to reboot:
`  
Setup ()
.......
stepperval = engine.stepperConnectToPin(stepPinStepper_V);
steppergorizont = engine.stepperConnectToPin(stepPinStepper_G);

pcnt_config_t pcntConf = { }; 
  pcntConf.unit = PCNT_UNIT_7; 
  pcntConf.pulse_gpio_num = metr_count;
  pcntConf.pos_mode = PCNT_COUNT_INC;  
  pcntConf.counter_h_lim = 10;       
  pcnt_unit_config(&pcntConf);        
  pcnt_counter_pause(PCNT_UNIT_7);      
  pcnt_counter_clear(PCNT_UNIT_7);     
  pcnt_set_filter_value(PCNT_UNIT_7, 1023); 
  pcnt_filter_enable(PCNT_UNIT_7);         
  pcnt_event_enable(PCNT_UNIT_7, PCNT_EVT_H_LIM);       
  pcnt_isr_register(summator, NULL, 0, NULL); 
  pcnt_intr_enable(PCNT_UNIT_7);                        
  pcnt_counter_resume(PCNT_UNIT_7);
  void loop()...
 `

gin66, 2025-01-20:
please try `pcnt_isr_handler_add()` instead of 
```
pcnt_isr_register(summator, NULL, 0, NULL);
pcnt_intr_enable(PCNT_UNIT_7);
```

diarus, 2025-01-20:
Thank you for your help. It is very necessary for beginners.
UINT_7 thinks fastaccelstepper is working correctly. But, a new problem has appeared: TG0WDT_SYS_RESET. See the code.  
TG0WDT_SYS_RESET is the reason for resetting the digital timer core of group 0 in Arduino ESP32.
What can help me?

`rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4`

gin66, 2025-01-20:
seems there is a missing watchdog reset somewhere in the code. something like `esp_task_wdt_reset();`.

see for reference the watchdog reset in FastAccelStepper:
https://github.com/gin66/FastAccelStepper/blob/27df2018ae4364776ad5a7fccefcd9351ffed7cd/src/StepperISR_esp32.cpp#L107

as I have no clue, which espidf is in use on your side, one possibility is to remove the #if/#endif in your local copy of FastAccelStepper.

It could be to anywhere else in the app, outside of FastAccelStepper.

gin66, 2025-01-21:
Does the reset happen with any of the examples e.g. UsageExample ?

gin66, 2025-02-20:
stale

closed 2025-02-20 by gin66