gin66/FastAccelStepper #309: Compilation problem, ESP32S3, Arduino Core 3.1.1, ESP-IDF 5.3.2

slmfd, 2025-02-12:
Hi! I'm trying to use this library with ESP32S3 in DRIVER_RMT mode, 
also i am using ESP32Encoder library which uses PCNT (this is why i'm trying DRIVER_RMT)
Sketch won't compile because even in DRIVER_RMT mode there are many PCNT defenitions in library, they are really needed?
So, i deleted all PCNT defenitions inside library and everything works ok, 
but which kind of problems i will face with deleting all PCNT defenitions?

gin66, 2025-02-12:
using rmt, then there is no issue to delete the pcnt stuff.
Could you please share part of the error log ?

slmfd, 2025-02-12:
if FastAccelStepper included first:

`In file included from C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/driver/deprecated/driver/pcnt.h:12:
C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/driver/deprecated/driver/pcnt_types_legacy.h:47:3: error: conflicting declaration 'typedef enum pcnt_unit_t pcnt_unit_t'
   47 | } pcnt_unit_t;
      |   ^~~~~~~~~~~
In file included from .pio/libdeps/lolin_s3/FastAccelStepper/src/fas_arch/common_esp32_idf5.h:124,
                 from .pio/libdeps/lolin_s3/FastAccelStepper/src/fas_arch/common_esp32.h:19,
                 from .pio/libdeps/lolin_s3/FastAccelStepper/src/fas_arch/arduino_esp32.h:11,
                 from .pio/libdeps/lolin_s3/FastAccelStepper/src/fas_arch/common.h:72,
                 from .pio/libdeps/lolin_s3/FastAccelStepper/src/FastAccelStepper.h:6,
                 from src/main.ino.cpp:16:
C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/esp_driver_pcnt/include/driver/pulse_cnt.h:21:16: note: previous declaration as 'struct pcnt_unit_t'
   21 | typedef struct pcnt_unit_t *pcnt_unit_handle_t;
      |                ^~~~~~~~~~~
C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/driver/deprecated/driver/pcnt_types_legacy.h:102:17: error: field 'unit' has incomplete type 'pcnt_unit_t'
  102 |     pcnt_unit_t unit;            /*!< PCNT unit number */
      |                 ^~~~
C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/esp_driver_pcnt/include/driver/pulse_cnt.h:21:16: note: forward declaration of 'struct pcnt_unit_t'
   21 | typedef struct pcnt_unit_t *pcnt_unit_handle_t;
      |                ^~~~~~~~~~~
.pio/libdeps/lolin_s3/ESP32Encoder/src/ESP32Encoder.h:62:21: error: field 'unit' has incomplete type 'pcnt_unit_t'
   62 |         pcnt_unit_t unit;
      |                     ^~~~
C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/esp_driver_pcnt/include/driver/pulse_cnt.h:21:16: note: forward declaration of 'struct pcnt_unit_t'
   21 | typedef struct pcnt_unit_t *pcnt_unit_handle_t;
      |                ^~~~~~~~~~~
Compiling .pio\build\lolin_s3\lib364\FastAccelStepper\FastAccelStepper.cpp.o
*** [.pio\build\lolin_s3\src\main.ino.cpp.o] Error 1`

And if ESP32Encoder included first:

`In file included from .pio/libdeps/lolin_s3/FastAccelStepper/src/fas_arch/common_esp32_idf5.h:124,
                 from .pio/libdeps/lolin_s3/FastAccelStepper/src/fas_arch/common_esp32.h:19,
                 from .pio/libdeps/lolin_s3/FastAccelStepper/src/fas_arch/arduino_esp32.h:11,
                 from .pio/libdeps/lolin_s3/FastAccelStepper/src/fas_arch/common.h:72,
                 from .pio/libdeps/lolin_s3/FastAccelStepper/src/FastAccelStepper.h:6,
                 from src/main.ino.cpp:16:
C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/esp_driver_pcnt/include/driver/pulse_cnt.h:21:16: error: using typedef-name 'pcnt_unit_t' after 'struct'
   21 | typedef struct pcnt_unit_t *pcnt_unit_handle_t;
      |                ^~~~~~~~~~~
In file included from C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/driver/deprecated/driver/pcnt.h:12:
C:/Users/User/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/include/driver/deprecated/driver/pcnt_types_legacy.h:47:3: note: 'pcnt_unit_t' has a previous declaration here
   47 | } pcnt_unit_t;
      |   ^~~~~~~~~~~
Compiling .pio\build\lolin_s3\lib364\FastAccelStepper\FastAccelStepper_idf4_esp32_pcnt.cpp.o
*** [.pio\build\lolin_s3\src\main.ino.cpp.o] Error 1`

gin66, 2025-02-12:
Two issues:

- ESP32Encoder is apparently using in espidf the legacy pcnt driver. This is not compatible with FastAccelStepper using the new driver.
- In the second run `FastAccelStepper_idf4_esp32_pcnt.cpp` is compiled, but with the conditional around, this should not happen. Perhaps the legacy stuff sets even the espidf version back.

Best would be, that ESP32Encoder gets upgraded.

slmfd, 2025-02-12:
Well, okay then, thanks for help!

closed 2025-02-12 by gin66