gin66/FastAccelStepper #310: ESP32 silent failure if initialising >2 motors

rqi14, 2025-02-28:
Hi. I found for ESP32 if I initialise more than 2 motors, it would freeze the board in the process of GPIO initialisation. It is completely fine with 1 or 2 motors.

```
I (765) gpio: GPIO[26]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (766) gpio: GPIO[16]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (773) gpio: GPIO[12]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (783) gpio: GPIO[25]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (792) gpio: GPIO[27]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (801) gpio: GPIO[12]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (811) gpio: GPIO[17]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pul�
```

gin66, 2025-03-01:
Sorry to hear that.

What comes to mind:
* If the third stepper is initialized first, will it fail with the first ?
* If the third stepper initialization is skipped and replaced by a delay for e.g. 1s, will if work ?
* Which version of espidf are you using ?
* esp32 or any derivate ?
* If espidf4: rmt or mcpwm/pcnt driver ?

There is a related issue #246, that in case of failure, FastAccelStepper still yield a stepper instance.

rqi14, 2025-03-03:
> Sorry to hear that.
>
> What comes to mind:
>
> * If the third stepper is initialized first, will it fail with the first ?
> * If the third stepper initialization is skipped and replaced by a delay for e.g. [...]

Hi. Thanks for the hints. It turned out to be a pin issue. The third motor is using pin 17 and 18. The output cuts at pin 17.
```
#define Z_DIR 01 // Z axis, direction pin
#define Z_STP 17 // Z axis, step pin
```
```
I (765) gpio: GPIO[26]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (766) gpio: GPIO[16]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (773) gpio: GPIO[12]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1783) gpio: GPIO[25]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1783) gpio: GPIO[27]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1791) gpio: GPIO[12]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2800) gpio: GPIO[19]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2800) gpio: GPIO[18]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2807) gpio: GPIO[12]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3817) gpio: GPIO[17]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pull�
```
From the build log, the esp-idf version is framework-espidf @ 3.40405.230623 (4.4.5)

The board is ESP32 ESP32. not subvariant.

I did not explicitly set rmt or mcpwm/pcnt driver. I am not sure what it uses by default.

gin66, 2025-03-03:
The esp32 board comes with psram ?

rqi14, 2025-03-03:
> The esp32 board comes with psram ?

Hi. The board I am using is called `ESP32 UNO D1 R32 CH340G`. It is not labelled to have PSRAM.

rqi14, 2025-03-03:
> The esp32 board comes with psram ?

Hi. I probably know why. Since Platform io does not have this board as an option, I selected `Adafruit ESP32 Feather` instead which seems to have PSRAM. Thanks for pointing it out. I will test later

closed 2025-03-03 by rqi14

rqi14, 2025-03-03:
> The esp32 board comes with psram ?

I tried `esp32doit-devkit-v1` and it did not solve the problem. maybe this board I bought had some issues. thanks anyway