gin66/FastAccelStepper #313: Inconsistency on using millis() and delay()

ricardohalla, 2025-03-21:
Hello everybody. I'm pretty new to Github, please let me know if this is not the right place to post this kind of stuff. Appologies in advance, if this is the case.

My problem is very simple, I have a closed-loop (with a built-in encoder) stepper motor and I was sucessful on controlling it by using FastAccelStepper. However, I was trying to do something using the `millis()` and `delay()` functions together with FastAccelStepper library. Then I face some problems. Here we go for the problem:

I have the following code:

```c
#include <FastAccelStepper.h>

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

const byte dirPinStepper = 3;
const byte enablePinStepper = 8;
const byte stepPinStepper = 6;

const long time = 5000;
int var = 1;

void setup() {
   Serial.begin(9600);
   engine.init();
   stepper = engine.stepperConnectToPin(stepPinStepper);
   if (stepper) {
      stepper->setDirectionPin(dirPinStepper);
      stepper->setEnablePin(enablePinStepper);
      stepper->setAutoEnable(true);
      stepper->setSpeedInHz(300);
      stepper->setAcceleration(150);
   }
}

void loop() {
   unsigned long currentMillis = millis();
   if (var == 1 && currentMillis > time) {
    Serial.println("Inside if statement");
    var = var + 1;
    stepper->move(-50);
   }
}
```
The motor should move to -50 one time and then nothing more happens, right? Well that didn't happen to me, what happen is when inside the "if statement" the motor emits a soft noise for a little bit and stops emitting this noise, no motion.

However, if we just replace `if (var == 1 && currentMillis > time) {` for `if (var == 1) {` the problem is fixed and the motor turn to -50 successfully. The same problem happens if I put a `delay()` in any part of the code, the motor doesn't turn. It is important to emphasize that the code does enter inside the "if statement" for both cases, I got this information due to the `Serial.println("Inside if statement");` line.

Here I bring some photos and the explanation of the wiring.

PUL+    ->    pin 6
DIR+    ->    pin 3
ENA+    ->    pin 8
PUL-, DIR- and ENA-    ->    GND

[image]
[image]
[image]

Can anyone help me? Why is this inconsistency happening?
Thank you!

gin66, 2025-03-21:
shouldn‘t it be `stepper->setEnablePin(enablePinStepper, false)` ?

ricardohalla, 2025-03-22:
> shouldn‘t it be `stepper->setEnablePin(enablePinStepper, false)` ?

I tried putting what you said:
`stepper->setEnablePin(enablePinStepper, false);`
And still didn't work.

However, changing `stepper->setAutoEnable(true);` for `stepper->setAutoEnable(false);` in addition to your suggestion (`stepper->setEnablePin(enablePinStepper, false);`), it worked properly!

I got to understand better what these functions do.
Thank you Mr. :)

closed 2025-03-22 by ricardohalla