gin66/FastAccelStepper #315: ESP32 - no position update, despite init success

judepereira, 2025-04-07:
hey! Firstly, thanks for this lib! I was able to test this using the supplied demo on my ESP32, and it worked well! However, integrating it into my larger project seems to be misbehaving, and I'm wondering what the next steps wrt debugging are.

The library is initialised in the same manner, with two steppers. When I invoke moveTo(), I know that something is queued internally. moveTo() returns MOVE_OK. I know that something is queued because isRunning() starts returning true (did not return true before moveTo() is invoked).

However, invoking **getCurrentPosition() in loop() always returns 0**.

At this point, I've double checked the call sequence, checked return values, and everything seems to be in place.

What could I do to debug this further?

gin66, 2025-04-07:
It's good to know that the base system is functioning, so the hardware seems to be fine. The question now is which other code is being used and which hardware modules it relies on. If there's a conflict, that might be causing the issue you're experiencing. 

Unfortunately, I can't determine which one without more details from your report. You could try disabling the initialization of other modules one by one to identify the culprit. Reordering the initialization sequence might also shed some light on the problem. Additionally, the ESP32's log file could provide helpful insights.

Wishing you the best of luck! If you can share more information about your system, I'll gladly assist further.

judepereira, 2025-04-08:
I managed to narrow it down to the WiFi management. When I moved the engine's init all the way up at firmware boot, I now see the following line:
```
[  2893][E][WiFiClient.cpp:429] write(): fail on fd 49, errno: 11, "No more processes"
```
That message appears to be printed over and over again. Have you seen something like this before?

gin66, 2025-04-08:
perhaps this has some pointers ?
(https://github.com/espressif/arduino-esp32/issues/6129)

judepereira, 2025-04-10:
> perhaps this has some pointers ?
Not a problem anymore. I removed most of the components, and the free stack is 14KB for loop(). I started debugging StepperTask.

I've narrowed it down to the following always being true, and not filling the queue:
```cpp
void FastAccelStepper::fill_queue() {
  // Check preconditions to be allowed to fill the queue
  if (!_rg.isRampGeneratorActive()) {
    return;
  }
...
}
```

judepereira, 2025-04-11:
Closing this since I made a silly mistake. I initialised my array of steppers incorrectly. I had `FastAccelStepper steppers[2];` instead of `FastAccelStepper *steppers[2];`.

Sorry about this.

closed 2025-04-11 by judepereira