gin66/FastAccelStepper #317: Support for SAMD51

loriiiiiiiii, 2025-04-18:
Hi
I'm currently using a SAMD51 microchip, which is not supported officially by fastaccelstepper. However, if I just change line 78 of `common.h` to
```
#elif defined(ARDUINO_ARCH_SAM) || defined(ARDUINO_ARCH_SAMD)
```
the code can be built. 

I'm wondering if this change is all that's needed for the support, or if a greater effort would be needed.

gin66, 2025-04-18:
Sounds promising, that there is no compile failure.
If you have the hardware, perhaps just give it a try?

the sam port has been done by @clazarowitz.
Maybe he has some feedback.

clazarowitz, 2025-04-18:
Honestly, I’d have to check the datasheet.  The pin mapping would be funny too I’m sure. You’re welcome to try it, but don’t expect much.  I’ll see if I can take a look at it later.  I’m actually back to working in a new stepper driver based on a tmc5160 and lmg1210 drivers for a big voltage/current range.  This project is going to kill me :D

gin66, 2026-09-04:
thanks to Lori Li <lihaoan1001@163.com> integrated the port in master

closed 2026-09-04 by gin66