gin66/FastAccelStepper #320: Stepper stalls for specific speed/acceleration/steps combinations

kpapadimakis, 2025-05-20:
First of all, thank you for this great library! I have already used it in one project of mine that helps driving a servo motor with a constant speed configured with an encoder and it works beautifully!

For my next one I have stumbled into a very weird situation though. I saw that for some specific values of speed / acceleration / steps to move, the stepper speed stalls. Changing a little one of the values results in the expected behavior and the motor goes where it was commanded. For example, changing the speed to 190000, or the acceleration to 900000 (or 1100000), or even by increasing or decreasing the target position by 1. The values that I see this behavior are shown in the code below.

I have tested and saw the same issue in 3 different ESP32S3 boards and made an example code that shows what I observe. I use Arduino framework in PlatformiO. The board I'm running this example is a Seeed Studio XIAO ESP32S3.

When the output is stalled, I can reset it by calling `forceStopAndNewPosition()`

Example code:
```
#include "FastAccelStepper.h"

#define dirPinStepper 3
#define enablePinStepper 5
#define stepPinStepper 6

#define stepperSpeed 200000
#define stepperAcceleration 1000000
#define stepperSteps 90000

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    delay(10);
  }

  delay(1000);
  Serial.println("Starting");

  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);

  stepper->setDirectionPin(dirPinStepper);
  stepper->setEnablePin(enablePinStepper);
  stepper->setAutoEnable(true);

  stepper->setSpeedInHz(stepperSpeed);
  stepper->setAcceleration(stepperAcceleration);
}

void loop() {
  int32_t stepper_position;
  static int32_t previous_position = 0;
  static unsigned long millis_stalled = 0;
  unsigned long stall_time = 0;
  static int direction = 1;

  if (stepper) {
    stepper_position = stepper->getCurrentPosition();
    if (stepper->isRunning()) {
      Serial.print("@");
      Serial.print(stepper_position);
      if (stepper_position == previous_position) {
        if (millis_stalled == 0) {
          millis_stalled = millis();
        }
        stall_time = millis() - millis_stalled;
        Serial.println(" (Stalled for " + String(stall_time) + "ms)");
        if (stall_time > 2000) {
          int32_t new_position = direction > 0 ? stepperSteps : 0;
          Serial.println("Stopping stepper and moving to " + String(new_position));
          stepper->forceStopAndNewPosition(new_position);
          millis_stalled = 0;
        }
        delay(1000);
      } else {
        Serial.println();
      }
      previous_position = stepper_position;
    } else {
      Serial.println();
      delay(1000);

      if (stepper_position == 0) {
        stepper->moveTo(stepperSteps);
        char buffer[50];
        snprintf(buffer, sizeof(buffer), "moveTo(%d)", stepperSteps);
        Serial.println(buffer);
        direction = 1;
      } else {
        stepper->moveTo(0);
        Serial.println("moveTo(0)");
        direction = -1;
      }
    }
  }

  delay(20);
}
```

Stall position was the same every time, although in my bigger project I have seen 2-3 different behaviors and a few times the motor did not stop at all

Output from my example:
```
moveTo(90000)
@230
@825
@1806
@3181
@5149
@7352
@9876
@12936
@16506
@20331
@24411
@28491
@32316
@36396
@40476
@44301
@48381
@52461
@56286
@60366
@64446
@68271
@70434
@70434 (Stalled for 0ms)
@70434 (Stalled for 1020ms)
@70434 (Stalled for 2040ms)
Stopping stepper and moving to 90000

moveTo(0)
@89770
@89175
@88194
@86819
@84851
@82648
@80124
@77064
@73494
@69669
@65589
@61509
@57684
@53604
@49524
@45699
@41619
@37539
@33714
@29634
@25554
@21729
@19566
@19566 (Stalled for 0ms)
@19566 (Stalled for 1020ms)
@19566 (Stalled for 2040ms)
Stopping stepper and moving to 0

moveTo(90000)
@230
@825
@1806
@3181
@5149
@7352
@9876
@12936
@16506
@20331
@24411
@28491
@32316
@36396
@40476
@44301
@48381
@52461
@56286
@60366
@64446
@68271
@70434
@70434 (Stalled for 0ms)
@70434 (Stalled for 1020ms)
@70434 (Stalled for 2040ms)
Stopping stepper and moving to 90000

moveTo(0)
@89770
@89175
@88194
@86819
@84851
@82648
@80124
@77064
@73494
@69669
@65589
@61509
@57684
@53604
@49524
@45699
@41619
@37539
@33714
@29634
@25554
@21729
@19566
@19566 (Stalled for 0ms)
@19566 (Stalled for 1020ms)
@19566 (Stalled for 2040ms)
Stopping stepper and moving to 0
```

gin66, 2025-05-20:
Thanks for the report and sorry for the inconvenience. I checked with an ESP32 because I don’t have an ESP32S3. Unfortunately, this setup doesn’t allow me to reproduce the stall. ESP32 and ESP32S3 have similar, but not identical, RMT drivers, which could be the reason.

On ESP32, I’ve observed that from around 168k Steps/s, the time to reach positions 0/90000 jumps roughly 1 second (slower). The stepper reaches reliably position 0 as confirmed by an attached pulse counter. However, I’m not sure what happens when the 90000 position is targeted. Since I don’t have a stepper connected, I need to check with a logic analyser or oscilloscope to see what’s happening.

Perhaps something similar is happening on ESP32S3, but not necessarily at the same speed.

gin66, 2025-05-21:
Could you please try out the latest version, if the situation is improved ?

kpapadimakis, 2025-05-22:
Thank you for looking at this so damn fast! I'll try and test it today

kpapadimakis, 2025-05-22:
And yes, it looks ok now!
Ran the code with 0.31.7 tag, which stalled repeatedly. After changing PlatformIO lib_deps to your master branch same code does not stall the motor, after running for 10'.

Thank you so much for the time and effort to fix this!

Greetings from Athens, Greece

gin66, 2025-05-22:
Thanks for the feedback and confirmation. 

Before release I need to perform some more tests on esp32 hw. Hopefully this weekend the next release will be available.

closed 2025-05-22 by gin66

gin66, 2025-05-22:
Good that I have not released the code. Thanks for sharing the updated findings.

Could you please provide the output of:
```
  Serial.print("ESP-IDF Version: ");
  Serial.println(esp_get_idf_version());
  Serial.printf("APB Base Clock: %d Hz\n", esp_clk_apb_freq());
```

In the meantime I have unified the rmt-code for espidf4.
Perhaps this new version is improved ?

gin66, 2025-05-22:
Promising results: my automated tests running on esp32 hw using rmt-driver have passed.
Curious about your results.

kpapadimakis, 2025-05-23:
I'm afraid nothing changed with your new commits. I see the end position values exactly as before (359923 and 77)

As for the prints you ask, I'm not very familiar with include paths etc and build system could not find esp32/clk.h, while clk.h was found in other paths which do not seem to be included.

```
/home/zaphod/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/soc/esp32/clk.h
/home/zaphod/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_hw_support/include/soc/esp32/clk.h
/home/zaphod/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s2/include/esp_hw_support/include/soc/esp32/clk.h
/home/zaphod/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/esp_hw_support/include/soc/esp32/clk.h

```

Found the function esp_clk_apb_freq in these files below though and so i included `<esp_private/esp_clk.h>` for testing if I got something from this

```
zaphod@eddie:~$ grep -R esp_clk_apb_freq ~/.platformio/packages/framework-arduinoespressif32
/home/zaphod/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/esp_hw_support/include/esp_private/esp_clk.h:int esp_clk_apb_freq(void);
/home/zaphod/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/esp_private/esp_clk.h:int esp_clk_apb_freq(void);
/home/zaphod/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_hw_support/include/esp_private/esp_clk.h:int esp_clk_apb_freq(void);
/home/zaphod/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s2/include/esp_hw_support/include/esp_private/esp_clk.h:int esp_clk_apb_freq(void);
```

So not sure if these values are the correct ones, but the results are these, if this helps:
```
ESP-IDF Version: v5.3.2-282-gcfea4f7c98-dirty
APB Base Clock: 80000000 Hz
```

gin66, 2025-05-23:
Thanks for,checking. This clarifies it and nothing else could be expected. The new changes are only done for espidf4. I still need to update for espidf5, too. The espidf major release change is impacting FastAccelStepper‘s rmt-driver, so adaptation is bit more work.

Stay tuned….

kpapadimakis, 2025-05-23:
I will! Thanks!

kpapadimakis, 2025-05-23:
Ok. also changed the platform to espidf4 and now it works ok. Also now I don't see the actual issue with the specific speed/acceleration/steps, so this seems to be the espidf5 all along!

I'll try downgrading my project to the espidf4 for now and will stay tuned for your espidf5 adaptation.

Thanks for everything!
Cheers

gin66, 2025-05-23:
New version with espidf4 and espidf5 versions unified is available.

Could please try again ?

kpapadimakis, 2025-05-24:
I tested just now and everything seems right with espidf5! 🎉 

You're fast
Thanks so much!