gin66/FastAccelStepper #321: Different behavior when using Arduino IDE and Platformio

julioghigi, 2025-05-29:
Hi! 

I compile and upload the following simple code using both Arduino IDE and Platformio to an ESP32-S3 from Waveshare:

`#include "FastAccelStepper.h"

#define dirPinStepper    37
#define enablePinStepper 6
#define stepPinStepper   36

// If using an AVR device use the definitons provided in AVRStepperPins
//    stepPinStepper1A
//
// or even shorter (for 2560 the correct pin on the chosen timer is selected):
//    stepPinStepperA

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
   engine.init();
   stepper = engine.stepperConnectToPin(stepPinStepper);
   if (stepper) {
      stepper->setDirectionPin(dirPinStepper);
      stepper->setEnablePin(enablePinStepper);
      stepper->setAutoEnable(true);

      stepper->setSpeedInHz(500);       // 500 steps/s
      stepper->setAcceleration(100);    // 100 steps/s²
   }
}

void loop() {
      stepper->moveTo(6400*4, true);
      delay(2000);
      stepper->moveTo(0, true);
      delay(2000);
}`

Since I am trying to control a JMC iHSV57, I monitored over RS232 the step count the motor was reporting and the strange behavior happened:

If I compile and upload over Arduino IDE, the JMC reports that receive all 25600 steps, and then go back to 0 in a loop.
If I compile and upload over Platformio, the JMC reports that receive only 6400 steps, then go back to 0.

Over Platformio is always a quarter of the desired steps and I have no idea why this happens. I imagine that is something over "platformio.ini" but I truly cant find what need to change.

Hope someone can help me!

gin66, 2025-05-29:
Thanks a bunch for the report! That’s quite an interesting behaviour, I must say.

I only use PlatformIO, so this is quite surprising. It’s quite unexpected that the PlatformIO version generates a quarter of the steps.

Which library version are you using? I’ve been making some changes to the rmt code recently, so I’d like to know if the issue is with the old or the new code. And what happens if we switch between the versions? If you could share the platformio.ini file, that would be fantastic!

julioghigi, 2025-05-30:
Hi! Thanks for the aswer! I was able to test just now.

Using `stepper = engine.stepperConnectToPin(stepPinStepper, DRIVER_RMT);` make it work flawless on the code.
Now compile and upload over Platformio the JMC motor reports all the 25600 steps.

I am using version "0.31.6" and this is my platformio.ini file:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.f_cpu = 240000000L
framework = arduino
lib_deps = 
	gin66/FastAccelStepper @ ^0.31.6
```

gin66, 2025-05-30:
Good to hear that the rmt driver works for you. The explanation for the different result is now clear. Arduino uses espidf5 under the hood and platformio still uses the older espidf4. Only espidf4 supports mcpwm/pcnt-driver, which is the default. While with espidf5 only the rmt-driver is provided. This means switching the development environment has switched the driver, which is not so obvious.

Unclear is why the mcpwm/pnt-combo is by a factor of 4 off. As the mcpwm/pcnt is a legacy driver, perhaps I just disabled it as unsupported!

BTW: You could give the latest 0.31.7 a try. This should contain high-speed rmt fixes. Or switch to the latest 0.33.0, which has some interface changes towards a more robust API.