gin66/FastAccelStepper #335: Motor does not move on ESP32 after update to 0.33.6 (works on 0.33.5)

kaito83, 2025-08-26:
Hi gin66!

I have some issue with new update.

After updating to version 0.33.6, the motor no longer moves on ESP32.
With version 0.33.5 everything works correctly.

I don’t see anything in the changelog that would indicate a breaking change for ESP32. The only change listed is:
0.33.6:
- pico2: support for PICO 2350 SDK (#334)

The motor movement is handled in my motor layer, which works on 0.33.5 but not on 0.33.6.

Here is my code

`#include "FastAccelStepper.h";
#include "steppers.h";
#include "EzDome.h";

//Stepper layer is responsible for the motor controlling

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *rotator = NULL;

//initing motor
void stepper::init(float speed, float accel) {
  engine.init();

  rotator = engine.stepperConnectToPin(rotator_stp);
  rotator->setDirectionPin(rotator_dir, rotator_dir_invert);
  rotator->setEnablePin(rotator_en, rotator_en_invert);
  rotator->setAutoEnable(true);
  rotator->setAcceleration(accel);
  rotator->setSpeedInHz(speed);
}

// rotor movement always find shortest route
long stepper::move(long move) {
  long current = rotator->getCurrentPosition();
  long delta = (move - current) % rot_full_rotation;

  if (delta > rot_full_rotation / 2) {
    delta -= rot_full_rotation;
  } else if (delta < -rot_full_rotation / 2) {
    delta += rot_full_rotation;
  }

  long target = current + delta;
  if (rot_zerosearch == true)
    rotator->moveTo(move);
  else
    rotator->moveTo(target);
  return delta;
}

//stoping the motor
void stepper::stop() {
  rotator->stopMove();
}

void stepper::forceStop() {
  rotator->forceStop();
}

//disable motors enable pin
void stepper::disable(bool S) {
  digitalWrite(rotator_en, S);
}

//updating motor position
void stepper::set_position(long pos) {
  rotator->setCurrentPosition(pos);
}

//query motor position
long stepper::get_position() {
  long pos;
  pos = rotator->getCurrentPosition();
  if (pos >= rot_full_rotation)
    pos -= rot_full_rotation;
  if (pos < 0)
    pos += rot_full_rotation;
  return pos;
}

//check motor is runing
bool stepper::isrun() {
  return rotator->isRunning();
}
`

gin66, 2025-08-26:
Sorry for the problem I have found the culprit and fixed the code. My esp32 hw tests have passed. Please give the new release a try.

kaito83, 2025-08-27:
Thank You, its fine now!

closed 2025-08-27 by kaito83